nanopb/nanopb #701: Comments generated incorrectly

SteveFrenchBOSS, 2021-09-14:


SteveFrenchBOSS, 2021-09-14:
Inside a message the comments generated for the fields follow the order the fields are listed.
The variables generated are place in order of the field number.

```
syntax = "proto2";
message order
{
    required uint32 a = 2; // comment for a
    required uint32 b = 1; // comment for b
}
```

generates

```
typedef struct _order { 
    uint32_t b; /* comment for a */
    uint32_t a; /* comment for b */
} order;
```

PetteriAimonen, 2021-09-14:
Hmm, interesting. Related to #85, #645.

As a workaround, one can consider setting option `option (nanopb_msgopt).sort_by_tag = false;` to keep the fields in `.proto` order.

PetteriAimonen, 2022-05-01:
Thanks for the report, this should be fixed now and be part of next release.

ewalt1, 2022-05-24:
This issue also applied when fields had been deprecated with type:FT_IGNORE, though ada8903102ca2e2fdf1c04e6f7d9e1a27fdcfe78 fixes that as well. Nice work!