nanopb/nanopb #702: int_size option in .options file results in: "pb_static_assertion_FIELDINFO_DOES_NOT_FIT_width2_field1_9_2" in 0.4.x

bboghrati, 2021-09-15:
**Steps to reproduce the issue**

With nanopb version 0.4.5 do:
1. define a message as: 
`message test {
    optional int32 progress = 1; 
}`
with `.options` file specifying:
`test.progress int_size:8`

2. Build 

**What happens?**
Build fails with:
`error: variably modified 'pb_static_assertion_FIELDINFO_DOES_NOT_FIT_width2_field1_9_2' at file scope [-Werror]`

Notes:
If the message is define as following builds fine:
`message test {
    optional int32 progress = 1 [(nanopb).int_size = IS_8]; 
}`

In [the tests](https://github.com/nanopb/nanopb/blob/master/tests/intsizes/intsizes.proto) the only syntax I could find uses options in the `.proto` file not a separate `.options` file. Is that suggesting that this is the preferred syntax and separate options files is being deprecated?

PetteriAimonen, 2021-09-15:
Nope, the two syntaxes are supposed to be equivalent and both supported.
I'm not getting the error you mention with GCC 9.3.0.

Can you take a diff out of the working and not working files and tell what is the difference?

bboghrati, 2021-09-15:
Thanks for the quick response. I tried this again in a smaller project and it works as intended. The issue I'm having occurs with both methods.

I'm closing this issue.

closed 2021-09-15 by bboghrati