nanopb/nanopb #703: istream errmsg not reported when set on decode callback

blumamir, 2021-09-15:
Hello,
I am decoding a message which has a field that is another message. 
I supply a callback for decoding the internal message. When this callback encounters an error, it sets `stream->errmsg = "some message"` and `return false;` to signal failure.

I was expecting to find this error message on the stream after the root pb_decode function returns, but instead, the property contains another message "callback failed".

Can it be related [to this part of the code](https://github.com/nanopb/nanopb/blob/ab19ecbe1b9f377ab4ee8e762bfe16c39068ad68/pb_decode.c#L763) not copying the errmsg from `substream` to `stream` before returning? the errmsg is copied in `pb_close_string_substream`, but it doesn't get executed in this flow.

Thanks for all the help

PetteriAimonen, 2021-09-15:
Yeah, it would make sense to retain the error message from the callback.
And the point of code you have identified is the one responsible.

PetteriAimonen, 2022-05-01:
Thanks for the report, this should be fixed now and be part of next release.

tr-3s, 2022-05-02:
I just encountered the same bug and found out that it was fixes yesterday.

Thanks @blumamir for finding and reporting and @PetteriAimonen for fixing it.

But I have a question here: Wouldn't it be better to call pb_close_string_substream() also in this case?

So maybe do

```
PB_SET_ERROR(substream, "callback failed"); // which retains the previous error on the substream, if any
pb_close_string_substream(stream, &substream);
return false;
```

This would deal with ```substream->bytes_left``` and ```stream->state``` more appropriately.

Or am I wrong here?

PetteriAimonen, 2022-05-02:
@tr-3s Good question. I'll have to consider it. It would do extra `pb_read()` unless the substream bytes left is set prior to calling `pb_close_string_substream()`.

closed 2022-05-30 by PetteriAimonen