nanopb/nanopb #705: pb_osteam_from_buffer: initializer element is not constant

matt-walker0, 2021-09-20:
Hi,

Trying to set up encoding on Raspberry Pi Pico, using GCC 9.2.1 arm-none-eabi, hit error fairly quickly on initialising pb_osteam.
Any ideas?

**Steps to reproduce the issue**
Compiler:
GCC 9.2.1 arm-none-eabi, both in VSCode and Make from terminal.

Code:
```
#include "protobuf/pb.h"
#include "protobuf/pb_common.h"
#include "protobuf/pb_encode.h"
#include "protobuf/my_proto.pb.h"
...
uint8_t protobuffer[200];
pb_ostream_t stream = pb_ostream_from_buffer(protobuffer, sizeof(protobuffer));

```

**What happens?**

Returns:
error: initializer element is not constant
[build]    13 | pb_ostream_t stream = pb_ostream_from_buffer(protobuffer, sizeof(protobuffer));

**What should happen?**

Shouldn't error :)

PetteriAimonen, 2021-09-21:
Try putting your stream inside a function, instead on in global scope.

matt-walker0, 2021-09-22:
> Try putting your stream inside a function, instead of in global scope.

Good shout, worked fine after that, cheers :)

closed 2021-09-22 by matt-walker0