nanopb/nanopb #706: Optional attributes' values are not cleared

YasmineSafi, 2021-09-23:
Hello, 

we are running a python application that receives json messages, parse it according to the given dict using `ParseDict` , serialize it `SerializeToString` and send it.

```python
# encode.py
            payload = ParseDict(data, data_t_v1())
            raw_payload = payload.SerializeToString()
```

This what we have in proto file:
```
//data.proto
syntax = "proto2";

message data_t_v1 {
	required uint32 id = 1;
	optional uint32 offset = 2;
	optional uint32 multiplier = 3;
	optional uint32 mask = 4;
}
```

However, I have an issue concerning the values of optional attributes. Here is what happens:
1- sending the message with the optional values

```python
data = {
   "id": 1,
   "offset": 2,
   "multiplier":10,
   "mask": 4294967295
}
```

2- sending the same message without the optional values
```python
data = {
   "id": 1
}
```
The issue happens here: `ParseDict` is returning a dict with the optional attributes having the old value that I set in step 1.

I would like to know if there is a way to clear old optional values when they are not set. 
Thanks in advance. and Please let me know if you need additional information

PetteriAimonen, 2021-09-23:
This does not sound like a nanopb issue. You may have better luck asking in whatever project the ParseDict function belongs to.

YasmineSafi, 2021-09-23:
@PetteriAimonen sure I will, Thanks for your quick reply!

closed 2021-10-21 by YasmineSafi