nanopb/nanopb #707: Add support to define options using precompiler symbols

labishrestha2, 2021-09-30:
Hi @PetteriAimonen! Thank you so much for creating this wonderful project! I've found it very useful and have leveraged it on several of my personal and professional projects. 

In one of my projects, I'd like to use the same protocol definition on several devices, but I'd like to allocate message/field storage space on a per-product basis. To achieve this, I'd like to add a way to define the message/field options using predefined symbols e.g.:

**MyMessages.proto**
```
message MyMessage {
      required string ID = 1;
}
```

**MyMessages.options**
```
MyMessage.ID                              max_size_symbol : "MY_MESSAGE_ID_SIZE"
```

Which would go on to generate:
```
typedef struct _MyMessage {
    char ID[MY_MESSAGE_ID_SIZE];
/* @@protoc_insertion_point(struct:MyMessage) */
} MyMessage;
```
and the message size as:
```
#define MyMessage_size                       (1 + MY_MESSAGE_ID_SIZE) // or whatever the correct size is
```

With this in place, I can define MY_MESSAGE_ID_SIZE outside of the generated folders on  per-project basis.

I wanted to open a issue report here to capture the feature and will create a new branch where I'll start adding such a feature. Please kindly let me know if there's already some way to achieve this. Thank you!

PetteriAimonen, 2021-10-01:
An alternate way is to use a separate `.options` file. Share the `.proto` between projects, but have individual `.options` in each project.

Regarding your suggestion, one risk is that users will define different values of `MY_MESSAGE_ID_SIZE` in different compilation units. But I guess as long as there is no default value, most users will figure out that it needs to be defined globally.

I'm nowadays extremely short on time that I can spend on nanopb, and the WIP patch seems quite extensive compared to the fairly rare usecase. Unfortunately I can't promise that I'll be able to review it in any reasonable timeframe.

labishrestha2, 2021-10-01:
Thank you for your consideration @PetteriAimonen! The separate `.options` way seems like a reasonable approach for now.
I'll keep tinkering with the code and leave the PR open in case anyone else would be interested to look into it - but feel free to remove it as you see fit. Regards!

pdgendt, 2024-01-27:
Hi @labishrestha2 & @PetteriAimonen 

I bumped into this issue, as this is just something I want! I use and maintain the `nanopb` module for the [zephyr rtos](https://docs.zephyrproject.org/latest/samples/modules/nanopb/README.html).

In the build system `Kconfig` is used to generate compile time constants which would be ideal for defining buffer sizes etc. as these are globally available.

For example:

```Kconfig
# Kconfig file
config SAMPLE_BUFFER_SIZE
	int "Simple message buffer size"
	default 8
	help
	  Configure the simple message buffer field's size.
```

```
// .options file 
SimpleMessage.buffer max_size:CONFIG_SAMPLE_BUFFER_SIZE fixed_length:true
```

```proto
// .proto file
syntax = "proto3";

message SimpleMessage {
    int32 lucky_number = 1;
    bytes buffer = 2;
}
```

The `Kconfig` constants allows the user have it configurable using `menuconfig` tools.

Is this a valid reason to continue with this effort? I'd like to help if needed.

PetteriAimonen, 2024-01-27:
@pdgendt Sounds reasonable, though I think you might consider preprocessing the .options file before generation to substitute the symbols there instead. I think even plain cpp might handle it.

pdgendt, 2024-01-27:
> @pdgendt Sounds reasonable, though I think you might consider preprocessing the .options file before generation to substitute the symbols there instead. I think even plain cpp might handle it.

I could create a cmake configure step, but it would be less descriptive for the resulting code. What do you mean with "plain cpp"?

PetteriAimonen, 2024-01-27:
By cpp I meant the C preprocessor.

But yeah, I do see the value in the feature. I just think that it will be difficult to implement in a way that is both useful and maintainable. The pull request #708 is a straightforward implementation, and is a reasonable approach. I think it has a few bugs and poor test coverage, but that could be improved. I also dislike the repetition between each option and its `_symbol` counterpart, but that may be unavoidable.

mastupristi, 2026-04-27:
> For what it's worth, how it now works in dev1.0 is that a precompiler symbol like `MyMessage_MyField_max_size` is used if size is not given. Then you can define that symbol in e.g. a custom header that can be included with `include` option.

I would like to add another use case for this feature request.

Currently, with a `.proto` like this:

```proto
message Prova
{
    bytes           prova  = 1;
    repeated uint32 prova1 = 2;
}
```

and an `.options` file like this:

```text
Prova.prova   fixed_length:true
Prova.prova1  fixed_count:true
```

I get a generated struct similar to:

```c
typedef struct Prova {
    pb_byte_t prova[Prova_prova_max_size];
    pb_size_t prova1_count;
    uint32_t prova1[Prova_prova1_max_count];
} Prova;
```

What I would like to have is a way to specify the array sizes using C preprocessor symbols, for example symbols generated by a build/configuration system:

```text
Prova.prova   max_size:CONFIG_MAX_SZ    fixed_length:true
Prova.prova1  max_count:CONFIG_MAX_CNT  fixed_count:true
```

and have these symbols emitted directly in the generated header, for example:

```c
typedef struct Prova {
    pb_byte_t prova[CONFIG_MAX_SZ];
    uint32_t prova1[CONFIG_MAX_CNT];
} Prova;
```

The important part is that `CONFIG_MAX_SZ` and `CONFIG_MAX_CNT` would be passed through verbatim to the generated `.pb.h`, instead of requiring nanopb to know their numeric values at generation time.

For the repeated field, when `fixed_count:true` is used together with a symbolic `max_count`, I would also expect the generated struct not to contain the `prova1_count` field, because the array size is fixed at compile time.

The motivation is to use the same configuration symbol consistently across the whole codebase. For example, a Kconfig-generated `CONFIG_*` symbol could define the limit once, and the nanopb-generated `.pb.h` would use that same symbol directly instead of requiring the value to be duplicated in the `.options` file.

I do not have a strong preference for the exact option syntax. Reusing `max_size` / `max_count` with symbolic values would be convenient, but dedicated options such as `max_size_symbol` / `max_count_symbol` would also solve the use case.