nanopb/nanopb #715: nanopb on IBMi

AlexeiBaranov, 2021-11-01:
Tried to port nanopb library to IBM i (x64 EBCDIC platform). Managed to compile sources and even tried to play with proto. 
I found an issue with the processing of repetitive data on IBM i. 
As far as I understand, with type:FT_POINTER option, you are placing the offset to field's *_count value in bits 28-31 of field descriptor. This limits the offset to *_count with values between 0 and 15, and it is not enough on IBM i.

Let's assume nanopb generated following structure in *.pb.h:
```
typedef struct _message { 
    pb_size_t str_count;
    struct _message_str *str; 
} message;
```

On IBM i,  all pointers are stored in structures with 16 alignment, although integer values may not be aligned. It means that offset(str) - offset(str_count) = 16, bits 28-31 of field descriptor becomes = 0: (((uint32_t)(16) & 0x0F) << 28)), and this leads to data decoding error "wrong size for fixed count field".

As a workarround, I'm using dummy variable:
```
typedef struct _message { 
    pb_size_t str_count_dummy;
    pb_size_t str_count;
    struct _message_str *str; 
} message;
```
It moves str_count 4 bytes away of structure beginning, leaving str on the same place. In this case offset(str) - offset(str_count) = 12, and "wrong size for fixed count field" is not raises. 

Is it possible to fix it in nanopb without such ugly workarround?

PetteriAimonen, 2021-11-01:
It's true that the 1-word and 2-word formats for field descriptor only has 4 bits for `size_offset` (https://github.com/nanopb/nanopb/blob/master/pb.h#L709).

If you are not very limited in code size, you can use the 4 word format which has 8 bit offset.

The preferred way to set this is via generator option `(nanopb).descriptorsize = DS_4`, which can be specified in either the `.proto` file, the `.options` file or on command line to `nanopb_generator`.

I wonder if there is some preprocessor macro that could tell about the platform minimum alignment size. Then it could be used to `#ifdef` out the 1 and 2 word formats in `pb.h`.

AlexeiBaranov, 2021-11-01:
Thank you, Petteri! Now it works without my dummy fields - I've used -s descriptorsize:DS_4 on command line. Lucky me, I'm not limited in code size (and actually in data size)

> I wonder if there is some preprocessor macro that could tell about the platform minimum alignment size. Then it could be used to #ifdef out the 1 and 2 word formats in pb.h.

On IBM i: 
>"All fields defined in ILE C/C++ structures are aligned on their natural boundaries. For example, int fields are four bytes long and are stored on four-byte boundaries". 

So minimum alignment here is 1 (single char values),  maximum - 16 (yes, we have 128 bit pointers). It wouldn't help to choose correct file descriptor format. I think 1,2,4 and 8 word formats are mostly related with code size limitations and projected data size. With 2 word format, you are limiting array sizes with 12-bit values (4096 elements), right?

PetteriAimonen, 2021-11-01:
Yeah, the formats are for saving code memory on embedded targets and are chosen automatically based on .proto info, but it does not know about C compiler specifics of course.

GCC apparently has `__BIGGEST_ALIGNMENT__`, but not sure if there exists anything portable for it.
C23 has `UINTPTR_WIDTH`.
Also `UINTPTR_MAX` exists, but it will be hard to test that when the C preprocessor only understands 32-bit values.
If there was, it could be used to automatically convert formats 1 and 2 to format 4 at C compiler time.

But maybe it is a bit rare to have platform with 128 bit pointers :)

AlexeiBaranov, 2021-11-01:
In any case, thank you for help. I'm closing the issue

closed 2021-11-01 by AlexeiBaranov