nanopb/nanopb #716: Data misalignment

jvieira88, 2021-11-17:
Hi,

I have a bit of a weird issue. After building a message and passing it by reference to another method, it seems like a portion of the data in memory is being skipped, resulting in data being misaligned. Adding a 16 bit field to the message seems to solve the issue.

Proto file:
```
syntax = "proto3";

import "MessageResponse.proto";
import "MessageVersion.proto";
import "MessageAlive.proto";

message WrapperMessage
{
	int32 iProtocolVersion=1;
	oneof msg{
		MessageResponse msgResp = 2;
		MessageVersion msgVersion = 3;
		MessageAlive msgAlive = 4;
	}
}
```

Message structure in WrapperMessage.pb.h:
```
/* Struct definitions */
typedef struct _WrapperMessage { 
    int32_t iProtocolVersion; 
    pb_size_t which_msg;
    union {
        MessageResponse msgResp;
        MessageVersion msgVersion;
        MessageAlive msgAlive;
    } msg; 
} WrapperMessage;
```

Building message:
```
void sendStatusMessage (void)
{
	WrapperMessage strMsg = WrapperMessage_init_default;

	strMsg.iProtocolVersion = 1;
	strMsg.which_msg = WrapperMessage_msgAlive_tag;
	strMsg.msg.msgAlive.bIsAlive = true;
	strMsg.msg.msgAlive.has_oTimeStamp = true;
	strMsg.msg.msgAlive.oTimeStamp.seconds = 0xAA;
	strMsg.msg.msgAlive.oTimeStamp.nanos = 0x55;

	enQueueTXMessage(&strMsg);
}
```

When debugging and looking at the memory contents, this is what I get:

Structure initialisation with some test data. All as expected here.
[image]

After passing the structure by reference, most of the data seems messed up. 
[image]

Looking at the memory contents, seems like all the data is there. However, the structure seems to be packed (no padding used) even though this wasn't specified in any of the structures defined in the *.pb.h files. 
[image]

Manually making the structures packed by adding a `#pragma pack(1)` on top of the structures definition seems to solve it.
Is there any other way of fixing this? Am I missing something?

For reference I'm using GCC.

PetteriAimonen, 2021-11-17:
So it seems that GCC is picking different layout for the same struct definition in different compilation units.
Try checking the arguments passed to GCC for the different files and make sure they have been recompiled since latest changes.

jvieira88, 2021-11-17:
Thanks for your comment Petteri. These are my arguments when compiling:
`-mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F411xE -c -I../Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -I../USB_DEVICE/App -I../USB_DEVICE/Target -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I"C:\Users\jv\source\repos\P1100_SonoTran\P1100_SonoTran/Core/CommsProtocol/NanoPB" -I"C:\Users\jv\source\repos\P1100_SonoTran\P1100_SonoTran\Core\CommsProtocol\CMessages" -O3 -ffunction-sections -fdata-sections -Wall -fstack-usage --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb`

Cleaned and recompiled, also tried different optimisation levels but with the same results.
In both cases (before and after passing the structure pointer), the data in memory looks like the same (packed, no padding). In memory, the structure members are not memory aligned, even though the structure definition in the .h files does not specify a packed structure...

PetteriAimonen, 2021-11-17:
You could try printing out `sizeof(WrapperMessage)` from different places, to see if the definition seen by the compiler is actually different, or if the debugger is just somehow showing it wrong.

jvieira88, 2021-11-17:
> You could try printing out `sizeof(WrapperMessage)` from different places, to see if the definition seen by the compiler is actually different, or if the debugger is just somehow showing it wrong.

The plot thickens. 
Before passing structure pointer. `sizeof(WrapperMessage) = 31`
After passing structure pointer. `sizeof(WrapperMessage) = 36`

I guess the pointer is using the correct definition of the structure (larger size) but I only have one definition of the `WrapperMessage`, so from where would the compiler be taking a different definition?

PetteriAimonen, 2021-11-17:
That's indeed curious. In the past I have seen this happen in a few situations:

- Multiple versions of the `foo.pb.h` in different directories, and include path working differently between files.
- One file C, other file C++, and different version of compiler used for C and C++.
- Different parameters given to compiler on command line (such as `-fpack-struct`)
- Old `.o` file that does not get recompiled for some reason.

It could also be that one of the header files being included in one of the `.c` files would use GCC pragma `#pragma pack()` to change the packing for all structures that come after it.

closed 2022-03-28 by PetteriAimonen