nanopb/nanopb #719: C++ support - nanopb_cpp

hacker-cb, 2021-12-07:
Hello @PetteriAimonen .

I made very good wrapper for nanopb with C++.

https://github.com/hacker-cb/nanopb_cpp

It can map all of the supported types (scalar types, maps, arrays, union messages, callbacks).
So, it very easy to reuse converters for inner messages. 

It has 100% tests with github actions  for all cases. (I hope)
Also it has good README and examples.

Could you please take a look on it?

I suggest to place `nanopb_cpp` repository to the `nanopb` organisation if you will satisfied my work. I will continue to maint it there.

A little bit later I will add platfromio support to it.

PetteriAimonen, 2021-12-08:
At a quick look this looks great and I'm willing to have the repo under nanopb organization!
I'll take a closer look and see how to go about the repo transfer in coming days.

Regarding `OUTPUT_STREAM_INCREASE_STEP`: any single value will not be enough as it would block long strings from being encoded. Instead set `stream->max_size` to `SIZE_MAX` initially. It would also be good to document that library user can and should set a lower limit to avoid filling all RAM in case of error.

hacker-cb, 2021-12-08:
Ok, done.

https://github.com/hacker-cb/nanopb_cpp/compare/bcc03409ef9ff7741dac93326bb079f1047ae678..6f9200970b0cc3c23bafce955c84ff22d0991538

PetteriAimonen, 2021-12-12:
Thanks. I've sent you an invitation to the nanopb organization and added permissions which should make it possible for you to transfer the repository according to [GitHub's help](https://docs.github.com/en/repositories/creating-and-managing-repositories/transferring-a-repository).

hacker-cb, 2021-12-12:
I moved and updated docs.

@PetteriAimonen Could you also add me to the nanopb organisation in platformio registry?

https://docs.platformio.org/en/latest/core/userguide/org/index.html#cmd-org

P.S. My Telegram is `hackercb` - so you contant me there for the future discussions.

PetteriAimonen, 2021-12-12:
@ivankravets Hmm, if I understood correctly the `nanopb` organization on PlatformIO is currently unmanaged and using something called "legacy sync". Can you help binding it to my username `petteriaimonen` on PlatformIO?

ivankravets, 2021-12-13:
Hi @PetteriAimonen,
Thanks for pinging me. You are added as the owner of `nanopb` organization in the PlatformIO Registry. I see there is legacy "autosync" enabled for your library, our crawler monitors "library.json" manifest and updated package in the registry. We recommend manually publish packages to the registry. See https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html#cmd-package-publish

```
pio package publish --owner nanopb
```

PetteriAimonen, 2021-12-13:
@ivankravets Thanks. I have it on my worklist to reorganize the nanopb CI flow. But for now it will be easier to keep using the library.json, as I'm very short on time lately.

@hacker-cb I added you to the organization on PlatformIO.

hacker-cb, 2021-12-13:
@PetteriAimonen thanks.
I will write workflow for publishing `nanopb_cpp`.
For the `nanopb` - I think we can reuse same workflow later.

hacker-cb, 2021-12-13:
@PetteriAimonen please:

1. Add organisation secrets:

`PLATFORMIO_USERNAME`
`PLATFORMIO_PASSWORD`

with your username/password for `pio account login` command. 
You can create separate pio account for the workflow, if you want...

2. Allow to use this secrets by the `nanopb_cpp` and `nanopb` repositories.

ivankravets, 2021-12-13:
@hacker-cb please don't export PlatformIO username/password for CI services. There is https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_AUTH_TOKEN

hacker-cb, 2021-12-13:
@ivankravets , ok thanks.

So, @PetteriAimonen , please set `PLATFORMIO_AUTH_TOKEN` secret in this case.

PetteriAimonen, 2021-12-13:
@hacker-cb I won't have more time to work on nanopb this year except for emergencies. You can open another issue for future CI improvements if needed.

closed 2021-12-13 by PetteriAimonen