nanopb/nanopb #725: Can't depend on nanopb library in PlatformIO unless I have PB files to compile

devalbo, 2021-12-11:
This is related to https://github.com/nanopb/nanopb/pull/723.

When I create a new PlatformIO project and add a dependency to nanopb, I am not able to compile using `pio run`. Exiting the `platformio_generator.py` process takes down the whole build, while reporting success. I am not a PlatformIO expert, but I suspect the module is Python imported into the PlatformIO build process, not running in its own process, and this exit takes down the whole build, not just the nanopb PB generation.

I like that I have the option for the nanopb integration to do my protoc compilation, but I don't like that I can't have the dependency without it. Adding a dependency to a project should not force me to change my build process unless there's a way to opt out.

To reproduce what I'm trying to prevent, perform the following steps:
* Create and navigate to a project directory - `mkdir proj && cd proj`
* Run `pio project init --board esp32dev`
* Add a file called `main.cpp` to `proj/src` with contents 
```
void setup() { }

void loop() { }
```
* Run 'pio run' and see the project will compile successfully
* Add nanopb dependency to project - `pio lib install nanopb/Nanopb@^0.4.6`
* Run 'pio run' and see the project will exit before compilation due to missing `nanopb_protos` (reporting success)

PetteriAimonen, 2021-12-11:
@hacker-cb

hacker-cb, 2021-12-11:
I will check.

hacker-cb, 2021-12-11:
Fixed and added tests for this case.

Problem was that I used `exit(0)` in script. 
I changed to `if/else`

hacker-cb, 2021-12-11:
@PetteriAimonen please publish update `0.4.7` to the platformio registry after merging PR.

closed 2021-12-11 by PetteriAimonen

PetteriAimonen, 2021-12-11:
@hacker-cb Hmm, those inflated version numbers in library.json are a bit problematic, even 0.4.6 doesn't really exist. I set it to 0.4.6.1 now, but in the future I guess something like "0.4.x-pre1" would be preferrable to keep the platformio versions in sync with the nanopb version numbers.

hacker-cb, 2021-12-11:
@PetteriAimonen 
I think it will be better to write github action which will increment version in `library.json` when new tag is created.
I have plans to do such action in `nanopb_cpp` project. Also I will make workflow to publish it to platfromio registry.

So, may be better to use workflow to publish `nanopb` to platformio?

PetteriAimonen, 2021-12-12:
I use `tools/set_version.sh`, which does set also platform.io version when I publish a new one. Platform.io is not the only distribution channel for nanopb, so the version numbers should be kept in sync.