nanopb/nanopb #731: Missing module in platformio library v 0.4.6

ekohlwey, 2021-12-18:
I receive an error when executing via pio run:
```
         *************************************************************
         *** Could not import the Google protobuf Python libraries ***
         *** Try installing package 'python3-protobuf' or similar.  ***
         *************************************************************
```

This can be remedied by running:

```
~/.platformio/penv/bin/pip3 install python3-protobuf
```

However it seems like this shouldn't be necessary, or should at least be described in the installation instructions if it is.

PetteriAimonen, 2021-12-18:
@prusnak Do you want to take a look at this? Maybe there is some platformio field that can be used to install the dependency.

The correct dependency names for pip are [protobuf](https://pypi.org/project/protobuf/) and [grpcio-tools](https://pypi.org/project/grpcio-tools/). The [python3-protobuf](https://pypi.org/project/python3-protobuf/) is some outdated non official package.

prusnak, 2021-12-18:
I am not very familiar with platformio.

@ekohlwey Can you please guide me how to reproduce the error? Ideally by sharing the minimalistic `platformio.ini` file?
Second question: why do you use 0.4.6 (unreleased) and not 0.4.5 (latest stable)?

PetteriAimonen, 2021-12-18:
@prusnak Ah sorry, wrong Pavel! I meant to ask @hacker-cb!

hacker-cb, 2021-12-18:
Currently you need to instal protobu on the build machine manually if you want to generate protos during platformio build:

 `apt install python3-protobuf protobuf-compiler`

I don't know how to add host dependencies to the platformio.

May be @ivankravets can point me to the right way how to add such dependencies?

PetteriAimonen, 2021-12-18:
This does something but I'm not sure it is the right way: https://github.com/platformio/platform-espressif32/blob/07919d0c5a731b06f01513766be0e72da38db313/builder/frameworks/espidf.py#L74-L97

ivankravets, 2021-12-18:
@PetteriAimonen yes, you are right. Just create [extraScript](https://docs.platformio.org/en/latest/librarymanager/config.html#extrascript) for the library, and use the next

`extra_script.py`
```
Import('env')

try:
    import  protobuf
except ImportError:
    env.Execute(
        env.VerboseAction(
            '$PYTHONEXE -m pip install  pip install protobuf',
            "Installing  Protocol Buffers dependencies",
        )
    )
```

hacker-cb, 2021-12-18:
@ivankravets thank you.

For the `pip install protobuf` its clear for me. but we also need `protoc` compiler. It is not a python package and have differnet pre-compiled binaries for different OS: https://grpc.io/docs/protoc-installation/

So, which is right way to install it in platformio?

PetteriAimonen, 2021-12-18:
@hacker-cb Actually protoc is in a Python package, grpcio-tools

hacker-cb, 2021-12-18:
@ivankravets thanks, you are right :)

@PetteriAimonen i wil make PR a little bit later.

hacker-cb, 2021-12-18:
I added grpc and protobuf [here](https://github.com/hacker-cb/nanopb/compare/platformio_protobuf_install..master)

But got error in [workflow run](https://github.com/hacker-cb/nanopb/runs/4569104314).

From the `nanopb` generator code seems that `invoke_protoc()` should use grpc-provided `protoc` but actually does not.

@PetteriAimonen - may be you can point me where to look for error?

```
[nanopb] Processing 'proto/pio_with_options.proto' (pio_with_options.options)
/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/../generator/protoc --nanopb_out=/home/runner/work/nanopb/nanopb/example/.pio/build/pio_with_options/nanopb/generated-src --nanopb_opt=--error-on-unmatched --proto_path=/home/runner/work/nanopb/nanopb/example/proto --nanopb_opt=-I/home/runner/work/nanopb/nanopb/example/proto pio_with_options.proto
Traceback (most recent call last):
File "/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/nanopb_generator.py", line 51, in <module>
 from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/../generator/protoc", line 6, in <module>
 from nanopb_generator import invoke_protoc
File "/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/nanopb_generator.py", line 72, in <module>
 import proto.nanopb_pb2 as nanopb_pb2
File "/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/proto/nanopb_pb2.py", line 23, in <module>
 create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
*** Error 1
[nanopb] ERROR: (1) processing cmd: '/home/runner/work/nanopb/nanopb/example/lib/nanopb/generator/../generator/protoc  --nanopb_out=/home/runner/work/nanopb/nanopb/example/.pio/build/pio_with_options/nanopb/generated-src --nanopb_opt=--error-on-unmatched --proto_path=/home/runner/work/nanopb/nanopb/example/proto --nanopb_opt=-I/home/runner/work/nanopb/nanopb/example/proto pio_with_options.proto'
```

hacker-cb, 2021-12-18:
Hm. seems that main error is `AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'`.

pip installs latest `grpcio-1.43.0` package. May be it is too fresh and need to install one of previous versions?

hacker-cb, 2021-12-18:
I tried `grpcio-tools===1.41.1` and `protobuf=3.17.3`, but got same error.

I have no more ideas.

hacker-cb, 2021-12-18:
It was my mistake, I specified single `=` instead of `==`.
With ptotobuf 3.19.1 all works fine.

PetteriAimonen, 2021-12-18:
Yeah, keeping python-protobuf and protoc versions in sync is sometimes a hassle.

Thanks for the pull-request, it is merged now. I also pushed new fake version 4.6.2 to PlatformIO (still haven't had time to study the newer publish mechanism).

closed 2021-12-18 by PetteriAimonen

hacker-cb, 2021-12-18:
@PetteriAimonen - I can make PR for auto-pushing to PlatformIO registry on creating release, like in [nanopb_cpp workflow](https://github.com/nanopb/nanopb_cpp/blob/master/.github/workflows/deploy-platformio.yaml).

General idea - version in `library.json` always `dev`. Workflow runs when making release and replace version in `library.json` to release tag (`git describe`).

I think it is not problem not increment patch version of the full nanopb release, event if changes was made only on PlatformIO or other non-main C headers/sources?

PetteriAimonen, 2021-12-18:
@hacker-cb Yeah, that's probably fine as the general idea. But currently nanopb releases are still in my own Jenkins instance & with my own scripting, so I prefer to keep what works for now.

hacker-cb, 2021-12-18:
@PetteriAimonen

Current `master` branch:

```
$ git describe 
nanopb-0.4.5-32-g8fbc844
```

I can write workflow which will use `0.4.5-32-g8fbc844` or `0.4.5-32` as version which you can run manually with worflow dispatch event? 

For the `library.json` - there are two options:
- use `dev` as version and replace it only in pio push workflow (without pushing to git). 
- replace `library.json` version and commit/push to master.

PetteriAimonen, 2021-12-18:
I won't be making any changes on that yet, as I don't know enough about the options and do not have time to consider them now.

hacker-cb, 2021-12-18:
Ok, let mw know when you will have time to discuss this.