nanopb/nanopb #734: Platform IO with platform espressif32 includes generated sources twice

BlackCrowned, 2021-12-20:
Platform IO includes generated source files twice when including nanopb in a project targetting espressif32.

**Steps to reproduce the issue**
* Create platformio with below platformio.ini
* Use nanopb version 0.4.6.2

```ini
[env:firebeetle32]
platform = espressif32
board = firebeetle32
framework = espidf
lib_deps = nanopb/Nanopb
nanopb_protos = +<protos/test.proto>
```

**What happens?**

Output of Build
```log
Warning! Ignore unknown configuration option `nanopb_protos` in section [env:firebeetle32]
Processing firebeetle32 (platform: espressif32; board: firebeetle32; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/firebeetle32.html
PLATFORM: Espressif 32 (3.4.0) > FireBeetle-ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40301.0 (4.3.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0) 
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r1
 - toolchain-xtensa-esp32 8.4.0+2021r1
 - toolchain-xtensa-esp32s2 8.4.0+2021r1
Reading CMake configuration...
Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in c:\users\user\.platformio\penv\lib\site-packages (3.19.1)
[nanopb] Skipping 'protos\test.proto' (no options)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in c:\users\user\.platformio\penv\lib\site-packages (3.19.1)
[nanopb] Skipping 'protos\test.proto' (no options)

*** Two environments with different actions were specified for the same target: C:\Users\user\temp\nanopb_sample\.pio\build\firebeetle32\nanopb\generated-build\test.pb.o
(action 1: xtensa-esp32-elf-gcc -o test.pb.o -c -DPLATFORMIO=50204 -DARDUINO_ESP32_DEV test.pb.c)
(action 2: xtensa-esp32-elf-gcc -o test.pb.o -c -Og -Wall -Werror=all -Wextra -Wno-error=deprecated-declarations -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wno-error=unused-variable -Wno-frame-address -Wno-old-style-declaration -Wno-sign-compare -Wno-unused-parameter -fdata-sections -ffunction-sections -fstrict-volatile-bitfields -ggdb -mlongcalls -std=gnu99 -D_GNU_SOURCE -DIDF_VER=\"4.3.1\" -DESP_PLATFORM -DPLATFORMIO=50204 -DARDUINO_ESP32_DEV test.pb.c)
File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 359, in <listcomp>
====================================================================================== [FAILED] Took 5.01 seconds ======================================================================================
The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

**What should happen?**
The generated sources should only be included once.

PetteriAimonen, 2021-12-20:
@hacker-cb Any ideas?

I don't immediately see why they would be included twice. And having two different sets of build options seems weird also.

hacker-cb, 2021-12-20:
@BlackCrowned 

As I see, your `platformio.ini` is not complete.

Could you make test project and provide its sources to reproduce bug?
You can attach zip here or create gist...

BlackCrowned, 2021-12-20:
Thanks for your quick response! 
This is the sample project. Can be opened with vscode and platformIO extension
[nanopb_sample.zip](https://github.com/nanopb/nanopb/files/7747334/nanopb_sample.zip)

hacker-cb, 2021-12-20:
Hm. I got same error when tried to compile your sources.

You are using `framework=espidf` and it generates additional CMake files in root and src folders. 

I don't know very wheel about `platformio` + `espidf` internals.

I just create `$BUILD_DIR/nanopb/generated_src_dir` in [platformio_generator.py](https://github.com/nanopb/nanopb/blob/master/generator/platformio_generator.py)
and add it it the builder with:

```
env.Append(CPPPATH=[generated_src_dir])
env.BuildSources(generated_build_dir, generated_src_dir)
```

My be @ivankravets can help us? 

As workaround, I replaced to `framework=arduino` and all works fine.

ivankravets, 2021-12-23:
@hacker-cb Your approach is good. Just one correction, please replace 
```python
nanopb_protos = env.GetProjectOption("nanopb_protos", "")
nanopb_plugin_options = env.GetProjectOption("nanopb_options", "")
```
with
```
nanopb_protos = env.GetProjectOption("custom_nanopb_protos", "")
nanopb_plugin_options = env.GetProjectOption("custom_nanopb_options", "")
```

To avoid warning from config parser. We validate all options and their types. Options that prefixed with `custom_` are user's options which we don't touch.

hacker-cb, 2021-12-24:
@ivankravets thank you. Fixed in GH-736.

What about `Two environments with different actions were specified for the same target` error with `framework=espidf` ?

May be it is some bug in PlatformIO wrapper for `espidf`?
@ivankravets , do you have any ideas?

ivankravets, 2021-12-24:
> Two environments with different actions were specified for the same target

Could you provide a link to the log?

hacker-cb, 2021-12-24:
@ivankravets log and test project archive were attached in this issue above by @BlackCrowned

ivankravets, 2021-12-27:
@valeros could you help here?

valeros, 2021-12-27:
Hi guys. Indeed, it's a peculiarity of the ESP-IDF integration. In a nutshell, we need to scan library dependencies two times.  And if the the second scan is performed as usual for any other framework, the first one is used to extract include paths in advance to be able to mix native ESP-IDF components, PlatformIO libraries and the project sources. It means that each time your extra script is executed as well, so `env.BuildSources` is called twice.

At the moment, the best workaround  that won't require any modifications in the framework build script is to use a special flag in the global build environment which detects that the script was already called and add the source files when the build environment is already contains all the necessary build flags. On the first call it would be great if you generate all the necessary header files and append all the include paths. On the second call you can add the sources. It might look something like this:
```python
    ....
    #
    # Add generated includes and sources to build environment
    #
    env.Append(CPPPATH=[generated_src_dir])

    global_env = DefaultEnvironment()
    if global_env.get("_PROTOBUF_SCRIPT_ALREADY_CALLED", False):
        env.BuildSources(generated_build_dir, generated_src_dir)
    global_env["_PROTOBUF_SCRIPT_ALREADY_CALLED"] = True

```

hacker-cb, 2021-12-27:
@valeros thank you for the idea. It works for single env.
I added `env['PIOENV']` to the end of the var name to make it work for multiply envs.

I made PR: GH-739

commit 74e3c796 (2022-01-11): PlatformIO: Fix ESP-IDF generation. (fix #734) (#739)
closed 2022-01-11 by PetteriAimonen