nanopb/nanopb #735: M_FLATTEN doesn't work

Xeenych, 2021-12-23:
I'm trying to set mangle_names option to M_FLATTEN
states.proto mangle_names:2 

but this results in exception
```
Traceback (most recent call last):
  File "nanopb_generator.py", line 2340, in <module>
  File "nanopb_generator.py", line 2228, in main_cli
  File "nanopb_generator.py", line 2158, in process_file
  File "nanopb_generator.py", line 1788, in generate_header
  File "nanopb_generator.py", line 1331, in fields_declaration
  File "nanopb_generator.py", line 1448, in default_value
AttributeError: 'Message' object has no attribute 'values'
[8044] Failed to execute script nanopb_generator
```

nanopb v.0.4.5 win

PetteriAimonen, 2021-12-23:
Please include the `.proto` file that reproduces the problem.

The M_FLATTEN is covered by this test case but it apparently doesn't trigger the problem you are having: https://github.com/nanopb/nanopb/blob/master/tests/typename_mangling/with_package.proto

Xeenych, 2021-12-23:
Renamed it to txt
[states.txt](https://github.com/nanopb/nanopb/files/7768894/states.txt)

PetteriAimonen, 2021-12-23:
The issue seems to be this message:

```
message OperationMode {
    enum OperationMode {
         White = 0;
         Color = 1;
         Scene = 2;
    }
    OperationMode operation_mode = 1;
}
```

The definition of `M_FLATTEN` is "Only use last path component". So in this case both the message and the enum would be called `OperationMode`, which won't work.

The error message could be better. The easiest fix is just to edit the proto to have different name for the enum.

Xeenych, 2021-12-24:
Ok. Got it. The message really could be better.

closed 2022-05-30 by PetteriAimonen