nanopb/nanopb #738: FEATURE: Resize pb_size_t accroding to max_count: option

Xeenych, 2021-12-24:
I have a repeated filed in a message.
I've set max_size option to 6.
I've got
```
    pb_size_t parameter_count;
    v1_states_Parameter parameter[6]; 
```
pb_size_t expands to unsigned short.
It would much better to make it uint8_t, so the struct could be packed more tightly

PetteriAimonen, 2021-12-24:
The tiny saving of RAM would be overwhelmed by the amount of extra code to handle different pb_size_t for each field.

closed 2021-12-24 by PetteriAimonen

Xeenych, 2021-12-24:
In my case this is the repeated field inside another repeated field, and it takes a huge amount of memory.

PetteriAimonen, 2021-12-24:
Usually it is better to use callbacks for large repeated fields, to avoid storing them completely in RAM

Xeenych, 2021-12-24:
I've tried to, but there is the problem
This is the example message:
```
message DeviceDesiredState {
     int64 timestamp = 1;
    //     .... "some other fields" here.
    repeated ChannelState channels = 2;
}
```
The host should check the timestamp and "some other fields" before executing a callback, or should pass these parameters to callback.
The order of fields in a serialized protobuf is unspecified.
Thus, I may receive 100 channels first and then timestamp and "some other fields". At the moment when I have a callback inside channels I have no knowledge about timestamp and "some other fields"