nanopb/nanopb #740: Native language comments in .options file

Xeenych, 2022-01-12:
I have comments in .options file in russian like:
```
// это комментарий
v1.states.ChannelState.parameter max_count:6
```

Nanopb cannot handle comments in native language in .options files
It produces the following error:
`UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 72: ordinal not in range(128)`

PetteriAimonen, 2022-01-12:
This appears to work correctly on Python 3, but gives the error message on Python 2.

I'm not sure if I'll get around to fixing it for Python 2, it is getting outdated anyway.

Xeenych, 2022-01-12:
I'm running nanopb_generator.exe not python

PetteriAimonen, 2022-01-12:
Hmm, which version?

Xeenych, 2022-01-12:
I'm using nanopb-0.4.5 linux binary from generator-bin/. .options file is in windows-1251 encoding.
Will try utf-8 in a moment

PetteriAimonen, 2022-01-12:
The 0.4.5 linux binary is working ok with UTF-8 for me. The 0.4.5 binaries are all made with Python 3 so should handle Unicode without issues.

Windows-1251 and other legacy non-UTF8 encodings will not be supported.

Xeenych, 2022-01-12:
Rechecked for UTF-8
```
[2841] Failed to execute script nanopb_generator
Traceback (most recent call last):
  File "nanopb_generator.py", line 2340, in <module>
  File "nanopb_generator.py", line 2223, in main_cli
  File "nanopb_generator.py", line 2098, in parse_file
  File "nanopb_generator.py", line 1946, in read_options_file
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd1 in position 72: ordinal not in range(128)
```

PetteriAimonen, 2022-01-12:
Hmm, apparently system locale may affect it.
Can you check what this command prints:

    python3 -c 'import locale; print(locale.getpreferredencoding(False))'

Xeenych, 2022-01-12:
I'm running this in docker container on Github actions:
````
python3 -c 'import locale; print(locale.getpreferredencoding(False))'
command not found
```

```
python -c 'import locale; print(locale.getpreferredencoding(False))'
ANSI_X3.4-1968
```

Xeenych, 2022-01-12:
Checked python --version
Python 2.7.18

We have python 2 in docker.

Will recheck with python 3 in a while

Xeenych, 2022-01-12:
This is with python 3
```
Traceback (most recent call last):
  File "nanopb_generator.py", line 2340, in <module>
  File "nanopb_generator.py", line 2223, in main_cli
  File "nanopb_generator.py", line 2098, in parse_file
  File "nanopb_generator.py", line 1946, in read_options_file
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd1 in position 3: ordinal not in range(128)
python3 -c 'import locale; print(locale.getpreferredencoding(False))'
UTF-8
locale
LANG=
LANGUAGE=
LC_CTYPE="POSIX"
LC_NUMERIC="POSIX"
LC_TIME="POSIX"
LC_COLLATE="POSIX"
LC_MONETARY="POSIX"
LC_MESSAGES="POSIX"
LC_PAPER="POSIX"
LC_NAME="POSIX"
LC_ADDRESS="POSIX"
LC_TELEPHONE="POSIX"
LC_MEASUREMENT="POSIX"
LC_IDENTIFICATION="POSIX"
LC_ALL=
python3 --version
Python 3.8.10
```

PetteriAimonen, 2022-01-12:
Hmm, I don't know why it works differently for you.
I don't currently have more time to debug this.

PetteriAimonen, 2022-03-28:
Hopefully this is now fixed, the encoding should be independent of system locale now.

commit 313069bf (2022-01-12): Fix unicode comment handling for Python 2 (fix #740)
closed 2022-05-30 by PetteriAimonen