nanopb/nanopb #741: nanopb_generate_cpp populates PROTO_SRCS with non-generated files

infn-ke, 2022-01-13:
When using `nanopb_generate_cpp`, this causes `PROTO_SRCS` to be populated with more than just generated files. This leads to a problem if also declaring these files as generated (as shown in the example snippets);

```
set_source_files_properties(${PROTO_SRCS} ${PROTO_HDRS}
  PROPERTIES GENERATED TRUE
)
```

Now doing a `ninja clean` will remove also non-generated files (e.g. pb_decode.c, pb_encode.c and pb_common.c.)

PetteriAimonen, 2022-01-14:
Hmm, if I understand correctly, it is caused by this line that appends nanopb source and headers to the list:

    set(${SRCS} ${${SRCS}} ${NANOPB_SRCS} PARENT_SCOPE)

But on the other hand to me it appears that the property you want is set already before that appending is done:

    set_source_files_properties(${${SRCS}} ${${HDRS}} PROPERTIES GENERATED TRUE)

In which case there would be no need to do that in your own CMake config.

infn-ke, 2022-01-14:
I was following this example here that explicitly sets this;

https://github.com/nanopb/nanopb/blob/master/examples/cmake_relpath/CMakeLists.txt#L13

But as you said the function `nanopb_generate_cpp` already sets this property here;

https://github.com/nanopb/nanopb/blob/master/extra/FindNanopb.cmake#L292

So you are right, no need to set it explicitly in my project.

PetteriAimonen, 2022-01-14:
Thanks, I have updated the examples!

infn-ke, 2022-01-14:
Thanks!

`nanopb_generate_cpp` generates a standalone library for every proto file. I think it would be better if the library linked against the static or shared library built when `nanopb_BUILD_RUNTIME` is set.

Maybe something like this (untested - need a way to inject the flags invoking the function)

```cmake
if (BUILD_STATIC_LIBS OR BUILD_SHARED_LIBS)
  set(${SRCS} ${${SRCS}} PARENT_SCOPE)
  set(${HDRS} ${${HDRS}} PARENT_SCOPE)
else()
  set(${SRCS} ${${SRCS}} ${NANOPB_SRCS} PARENT_SCOPE)
  set(${HDRS} ${${HDRS}} ${NANOPB_HDRS} PARENT_SCOPE)
endif()
```

infn-ke, 2022-01-14:
Yeah, need to find a way to to not break backward compatibility. So maybe like this then;

```cmake
if(NOT DEFINED NANOPB_GENERATE_CPP_STANDALONE)
  set(NANOPB_GENERATE_CPP_STANDALONE TRUE)
endif()

if (NANOPB_GENERATE_CPP_STANDALONE)
  set(${SRCS} ${${SRCS}} ${NANOPB_SRCS} PARENT_SCOPE)
  set(${HDRS} ${${HDRS}} ${NANOPB_HDRS} PARENT_SCOPE)
else()
  set(${SRCS} ${${SRCS}} PARENT_SCOPE)
  set(${HDRS} ${${HDRS}} PARENT_SCOPE)
endif()
```

PetteriAimonen, 2022-01-14:
Yeah, that looks good! I added it.

closed 2022-01-14 by infn-ke