nanopb/nanopb #742: AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'

EnziinSystem, 2022-01-14:
My environment:

* Python 3.8.10
* Ubuntu 20.04

I installed: `pip install protobuf grpcio-tools`

When I run the command:

```
python ../nanopb/generator/nanopb_generator.py network_protocol.proto
```

The error message:
```
Traceback (most recent call last):
  File "../nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/kevin/Desktop/Working/Protocol/nanopb/generator/proto/nanopb_pb2.py", line 23, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
```

PetteriAimonen, 2022-01-14:
That is probably caused by system python-protobuf version changing (#733).

Try removing the autogenerated file `generator/proto/nanopb_pb2.py`.
It will be regenerated automatically and hopefully with correct version.

PetteriAimonen, 2022-02-17:
Actually this happens if the system already has python-protobuf installed, but it is of an old version. The grpcio-tools package didn't specify the minimum version it requires (https://github.com/grpc/grpc/issues/28479 ).

As a workaround one can specify `--upgrade` to make sure pip installs newest versions:

    pip install --upgrade protobuf grpcio-tools

closed 2022-05-30 by PetteriAimonen