nanopb/nanopb #743: Howto traverse all callbacks? Howto handle variable length fields?

infn-ke, 2022-01-14:
Is there a way to traverse all callbacks setup (other than invoking `pb_encode`)? 

Is it considered an anti-pattern to allocate memory for deep copy when callbacks are setup and free that same memory when the callback is invoked as a result of `pb_encode`? Any better design pattern for handling variable length? Considering an options.proto with fixed size as a last resort.

PetteriAimonen, 2022-01-14:
Well, fixed size is often the correct approach for embedded systems. When memory is limited anyway, failing `malloc()` is much harder to handle correctly and reliably than a per-field maximum size.

Callbacks are really meant for when the data is going somewhere else than just RAM. If you are just going to store them in RAM with dynamic allocation, you could use `PB_ENABLE_MALLOC`.

infn-ke, 2022-01-14:
`PB_ENABLE_MALLOC` only helps when decoding messages? Anything similar for encoding messages? I have a oneway interface, so only interested in the encoder part.

PetteriAimonen, 2022-01-14:
For encoding, you can specify `(nanopb).type = FT_POINTER` even without having `PB_ENABLE_MALLOC` enabled. Then you get pointers for each field, which you can point to the actual data.

If you do use `PB_ENABLE_MALLOC`, then you can call `pb_release()` and it will call the function defined as `pb_free()` on each pointer. This can be helpful if you allocate your data to be encoded using `malloc()`.

infn-ke, 2022-01-17:
Thanks for the pointers! I'll have a look.

The options proto file is looked up in the same path as the proto file being compiled. Can you also inject options files as input to the `nanopb_generate_cpp`?

PetteriAimonen, 2022-01-17:
Yeah, the `FindNanopb.cmake` rules should automatically detect `.options` file in same directory. You can also set `NANOPB_OPTIONS_DIRS` if they are elsewhere.

closed 2022-01-17 by infn-ke