nanopb/nanopb #744: decode bytes using callback in oneof

pietercardoen, 2022-01-21:
I have a protobuf file with a value of type oneof:
```
        oneof value {
            ...
            bytes    bytes_value                    = 16;       // Bytes, File
            ...
            MetricValueExtension extension_value    = 19;
        }
```

I am decoding the MetricValueExtension using message level callbacks for oneofs. I am succeeding in doing so. However when I am trying to decode the bytes_value using a callback, I'm facing issues.

The documentation states: "When a field inside oneof contains pb_callback_t fields, the callback values cannot be set before decoding. This is because the different fields share the same storage space in C union. Instead either function name bound callbacks or a separate message level callback can be used. See tests/oneof_callback for an example on this."

For submessages, the callback cb_value is called but when decoding bytes it isn't.

How do I need to solve this?

PetteriAimonen, 2022-01-21:
Hmm, indeed the message level callback is currently only implemented for submessages. I'm not sure I want to extend that, it was a bit cumbersome workaround for the oneof+callback issue in the first place.

Name bound callbacks should work: set `(nanopb_msgopt).callback_function = "mydecodecallback"` and then implement a function with that name. It will get called for all callback fields inside the message, instead of the normal callback mechanism. It also has the benefit that you can have custom datatypes inside the structure instead of `pb_callback_t`. See e.g. https://github.com/nanopb/nanopb/blob/master/examples/network_server/client.c#L30 for example.

If that does not suit your usecase, another workaround could be to use generator option `(nanopb).no_unions = true`, which generates the fields separately and callbacks can be set directly.

pietercardoen, 2022-01-24:
Thanks for the reply.

I've solved it now by using FT_POINTER instead of FT_CALLBACK.

closed 2022-01-24 by pietercardoen