nanopb/nanopb #745: pb_decode_raw callback (like protoc --decode_raw)

maximevince, 2022-01-25:
It would seem useful to me to have an interface that allows you to decode any protobuf byte array, and receive a callback on every new field found. 
The callback could report the field number and it's wire type.
This would allow the user of the callback to act appropriately depending on the field number and it's wire type.
Another advantage is that no huge RAM buffers are needed to store the entire message struct. The message can be parsed as-you-go, and values can be consumed where needed.

Example callback signature (very similar to the field-specific callback interface now):
```C
bool ( * pb_decode_raw_cb) (pb_istream_t *stream, pb_wire_type_t wire_type, pb_field_iter_t *field)
```

My specific use-case contains a big protobuf message where all fields are optional. Depending on the fields included, the firmware would apply some of these fields as new settings, or execute a specific command.
However, because it contains hundreds of optionals, the required RAM size is rather big (> 1500 bytes).

This is also the reason the field-specific callback interface does not suit my use-case very well, since this would require me to assign 200+ callbacks / function pointers for each and every field number in my message definition.

Another advantage of this approach, is that even if the `.proto` definition is not available, a raw protobuf struct can still be decoded.

Is this already possible somehow, and am I just missing how to accomplish this?
Or would you agree that this is useful, and give me some pointer on how to implement this?
I am willing to implement this functionality, and send a pull request.

Thanks for the nice project!

PetteriAimonen, 2022-01-25:
You can just call `pb_decode_tag()`.

See e.g. [tests/raw_decode](https://github.com/nanopb/nanopb/tree/master/tests/raw_decode)

maximevince, 2022-01-25:
Oh no, I totally missed that, and even the given example. Awesome, that solves my case entirely.
Thanks again!

closed 2022-01-25 by maximevince

PetteriAimonen, 2022-01-25:
You're welcome :)

As a future reference, the signature for the name-bound callbacks is actually exactly what you suggest:
https://github.com/nanopb/nanopb/blob/master/examples/network_server/client.c#L30

But even if one set `(nanopb).callback_datatype="struct {}"` to make no-storage fields, it would still needlessly store the descriptor for each field, so not very useful for your use case.