nanopb/nanopb #746: problem generating with nested messages and enums

ceggers-arri, 2022-01-31:
Observed with version 0.4.3 and 0.4.5. Probably related to #487.

After adding a new enum, the nanopb generator raises an error. Without the enum member, no error occurs.

**file.proto:**
```
syntax = "proto2";  // must be "proto2", proto3 requires enums starting with 0.

enum Enum
{
	enumerand = 1;  // must be non-zero
}

message Outer1
{
	message Inner
	{
		optional Enum enum1 = 1;  // can also be "required"
	}
}

message Outer2
{
	message Inner
	{
		// further trouble of this is also named "enum1"
		optional Enum enum2 = 1;  // can also be "required"

	}
}
```
**Error messages** (file paths shortened):

> Traceback (most recent call last):
>   File ".../nanopb/generator/nanopb_generator.py", line 2151, in <module>
>     main_plugin()
>   File ".../nanopb/generator/nanopb_generator.py", line 2136, in main_plugin
>     results = process_file(filename, fdesc, options, other_files)
>   File ".../nanopb/generator/nanopb_generator.py", line 1984, in process_file
>     headerdata = ''.join(f.generate_header(includes, headerbasename, options))
>   File ".../nanopb/generator/nanopb_generator.py", line 1631, in generate_header
>     yield msg.fields_declaration(self.dependencies) + '\n'
>   File ".../nanopb/generator/nanopb_generator.py", line 1186, in fields_declaration
>     defval = self.default_value(dependencies)
>   File ".../nanopb/generator/nanopb_generator.py", line 1316, in default_value
>     desc = google.protobuf.descriptor.MakeDescriptor(optional_only)
>   File ".../usr/lib/python3.8/site-packages/google/protobuf/descriptor.py", line 1077, in MakeDescriptor
>     _message.default_pool.Add(file_descriptor_proto)
> TypeError: Couldn't build proto file into descriptor pool!
> Invalid proto descriptor for file "4961cb88fa20c55bf8be6a2eb19c2f55.proto":
>   Inner: "Inner" is already defined in file "f6ee16dcd01f8fe659d50ee18fcef754.proto".

A nanopb options file was used, but it was empty.

**What Google says:**
- https://github.com/ValvePython/csgo/issues/8
- https://stackoverflow.com/questions/68952308/tensorflow-couldnt-build-proto-file-into-descriptor-pool

Following these instruction and building python3-protobuf without `--cpp_implementation` or using `pip` with `--no-binary` seems to avoid this problem.

PetteriAimonen, 2022-03-28:
Thanks for the report!

Should be fixed now, by adding an unique id to the submessage name when using it for the default value generation.

ceggers-arri, 2022-03-29:
Thanks for working on my bug report.

I had some difficulties in reproducing the original problem. I use nanopb together with the OpenEmbedded/Yocto SDK and upgraded recently to a newer version (which also includes a more recent version of python3).

- Yocto **Kirkstone** (current dev snapshot, Python **3.10.2**):
  * I cannot reproduce the original problem (even after reenabling `--cpp_implementation` for python3-protobuf).
- Yocto **Dunfell** (older version from my CI, Python **3.8.11**):
  * I can reproduce the original problem with nanopb-0.4.5.
  * I can reproduce the original problem with the latest nanopb head after reverting 6146127.
  * The problem is **solved** with 6146127.

closed 2022-03-29 by ceggers-arri