nanopb/nanopb #747: raw_decode parsing of strings differ from protoc --decode_raw

maximevince, 2022-02-04:
Hi,

Working with `raw_decode` from https://github.com/nanopb/nanopb/tree/master/tests/raw_decode works really well for VARINT/64BIT/32BIT types. However, when trying this on a String type, it seems to interpret this in the wrong (different?) way.

Sample payload: `482A788901F8040180050CDA10096C616C61616C616C61`
Using `protoc --decode_raw`, I get:
```
$ echo -n 482A788901F8040180050CDA10096C616C61616C616C61 | xxd -r -p | protoc --decode_raw

9: 42
15: 137
79: 1
80: 12
267: "lalaalala"
```

Using `raw_decode`, I get:
```
$ echo -n 482A788901F8040180050CDA10096C616C61616C616C61 | xxd -r -p | ./rawdecode

At 0: field tag 9, wire type 0 (VARINT), varint value (1 bytes): 42
At 2: field tag 15, wire type 0 (VARINT), varint value (2 bytes): 137
At 5: field tag 79, wire type 0 (VARINT), varint value (1 bytes): 1
At 8: field tag 80, wire type 0 (VARINT), varint value (1 bytes): 12
At 11: field tag 267, wire type 2 (STRING), string len 9 bytes, attempting recursive decode
f267> At 14: field tag 13, wire type 4 (EGRP)
f267> At 15: field tag 12, wire type 1 (64BIT)
f267> ERROR: Failed to parse fixed64: end-of-stream
LATEST BYTES READ (16 to 16): 
field 267: recursive decode failed, continuing with upper level
```

Maybe, I am misinterpreting how `raw_decode` is supposed to handle string?

Thanks for you help!

PetteriAimonen, 2022-02-04:
In protobuf format, strings and submessages use the same wire type 2, so without a `.proto` file one cannot know if it is a submessage.

The `raw_decode` is mainly meant for debugging so it tries to decode strings as submessages, and skips them if it cannot do so. This is useful to debug corruption inside submessages, but not very useful for seeing text.

maximevince, 2022-02-04:
Hi @PetteriAimonen,

I did not know it worked that way. That explains the raw_decode output.

Thanks for the incredibly fast reply again!

closed 2022-02-04 by maximevince