nanopb/nanopb #748: sub message not encoded

kenjichanhkg, 2022-02-04:
I have a proto file like below:

```
syntax = "proto3";
message GeoPoint {
    double lon  = 1;
    double lat  = 2;
}

message BuoyUpdate {
    GeoPoint pos = 1;
    double another_field = 2;
}

```

Then I encode it:
```
BuoyUpdate msg = BuoyUpdate_init_zero;
msg.pos.lat = 12.0;
msg.pos.lon = 23.0;

uint8_t buffer[BuoyUpdate_size] = { 0 };
pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
auto status = pb_encode(&stream, BuoyUpdate_fields, &msg);
if (!status) {
    fmt::print("failed to encode BuoyUpdate, {}\n", PB_GET_ERROR(&stream));
    return;
}
fmt::print("protobuf bytes_written = {}\n", stream.bytes_written);

```

Now the problem is, stream.bytes_written is always zero. It's unable to encode the pos of GeoPoint sub message.
I tried to assign an value to another_field, it works. So I assume something is wrong with sub message.

Can anyone help me please?

PetteriAimonen, 2022-02-05:
Which nanopb version are you using? This sounds a bit like bug #504 which was fixed in 0.3.9.6 & 0.4.2.

Also, which processor & compiler? Does it work if the submessage contains e.g. int32 instead of double?

PetteriAimonen, 2022-02-05:
Try setting:

     BuoyUpdate.has_pos = true;

While in proto3 normal fields are present when non-zero, submessage fields have explicit present/not present status.

(And yeah, this is somewhat confusing, but that appears to be the way the protobuf standard is intended to be.)

kenjichanhkg, 2022-02-05:
yeah thanks!  has_pos = true.  works!
it'd be nice if has_pos is set automatically, but currently not possible, unless people must use set_pos(), and use pos() for reading

PetteriAimonen, 2022-02-05:
And nanopb is a C library anyway, so no setters.

There is option `proto3_singular_msgs` that makes messages present if 
any of their inner fields is set. That was the default until nanopb 
0.4.0, but it turned out to be my misinterpretation of proto3 spec. It 
can still be used if preferred, though.

closed 2022-03-28 by PetteriAimonen