nanopb/nanopb #749: Wrong output directory

FeUs17, 2022-02-10:
I'm running the following command

`
nanopb_generator.py --output-dir=src/msg/auto_generated src/msg/protos/my_msg.proto
`

I've got two problems:
1. There is not a --proto_path flag
2. the two generated files (my_msg.pb.c and my_msg.pb.h) are created in the following directory:
   src/msg/auto_generated/src/msg/protos/

In other words, the full name of the proto file, including the path, is added to the output_dir.
This would be easily fixed adding 

`
    noext = os.path.basename(noext)
`

in the process_file function, at line 2159, nanopb-0.4.6-dev, after:
`
noext = os.path.splitext(filename)[0]
`

I admit that the usage of the flag would be much better, but I think this is an easy fix that should be done anyway.

PetteriAimonen, 2022-02-10:
This is the way Google's `protoc` works.

Regarding `--proto-path`, there is `-I`. This command may be what you want:

    python ~/nanopb/generator/nanopb_generator.py \
                  -Isrc/msg/protos/  \
                  --output-dir=src/msg/auto_generated \
                  src/msg/protos/my_msg.proto
    Writing to src/msg/auto_generated/my_msg.pb.h and src/msg/auto_generated/my_msg.pb.c

woodz-, 2022-03-15:
But then your -I option is poorly documented. If you type

    $ generator/nanopb_generator.py --help

you get:

    ...
    -I DIR, --options-path=DIR
                        Search for .options files additionally in this path
    ...

The fact that `-I` is also responsible to propagate `--proto-path` against the `protoc` conmpiler is not stated there. It might be deep in one of the docs anywhere...

PetteriAimonen, 2022-03-15:
I agree. Sorry to say, but "pull requests welcome"; my time and interest is pretty limited..

closed 2022-03-28 by PetteriAimonen