nanopb/nanopb #750: feature request: support unknown fields

Kxuan, 2022-02-11:
The unknown fields are useful to passing messages to another service.

Currently, when a field is not found in decoding messages, nanopb just skip the field.
To support unknown fields, nanopb can callback to user. And nanopb pass the field_number, wire_type, and raw field data to the user, then the user store them in some way.

To encoding messages, the user provides the field_number, wire_type, and raw field data back to nanopb. Then nanopb encode them into streams.

Links:
https://developers.google.com/protocol-buffers/docs/proto3#unknowns

PetteriAimonen, 2022-02-11:
One can do something similar in two ways currently:

* Decode/encode tags manually, see `tests/raw_decode`
* Register custom extension handler, `pb_extension_t`. The callback gets tag and wire type for all unknown fields. Currently that requires that the message is declared as extensible in `.proto` file, but a nanopb option could be added for that also.

closed 2022-05-30 by PetteriAimonen

schneider42, 2023-03-08:
Is my understanding correct that `message is declared as extensible` means that it has an extension range defined?

Extension ranges are not supported in proto3 and I'm wondering: is there is a way to achieve the what the extension handler does with proto3?