nanopb/nanopb #754: Generated proto2 structs require large amounts of padding

mark64, 2022-02-20:
Nanopb structures for proto2 messages have a pattern:
```
struct X {
  bool has_y;
  // Implicit padding on most platforms of 3 bytes
  int32_t y;
  bool has_z;
  // Implicit padding on most platforms of 3 bytes
  uint64_t z;
};
```
Were the structures laid out as below, we'd save 6 / 20 bytes, a 30% reduction in nanopb struct memory requires in the above example!
```
struct X {
  int32_t y;
  uint64_t z;
  bool has_z;
  bool has_y;
};
```

I tried to implement that with the following code, before realizing that there's a lot of nanopb generation code that depends on the ordering of struct fields. I suspect this would require a large refactor of nanopb generation to happen, but curious if you have ideas on how this could be done, or reasons that it shouldn't?

```
diff --git a/generator/nanopb_generator.py b/generator/nanopb_generator.py
index 1291c88..d4d1408 100755
--- a/generator/nanopb_generator.py
+++ b/generator/nanopb_generator.py
@@ -1218,7 +1218,28 @@ class Message(ProtoElement):
             # Therefore add a dummy field if an empty message occurs.
             result += '    char dummy_field;'
 
-        result += '\n'.join([str(f) for f in self.fields])
+        # To optimize the size required to store the struct in memory,
+        # we sort the generated lines so that smaller fields are the end.
+        # This reduces the need to insert empty padding.
+        # https://stackoverflow.com/questions/4306186/structure-padding-and-packing
+        contents_lines = [str(f) for f in self.fields]
+        # Instead of actually sorting the struct's fields by size, we split
+        # the lines into those that are bool typed and those that are not.
+        # While not perfect, this does get us very close to optimal packing
+        # on proto2 messages (which have lots of bool has_X fields between
+        # fields).
+        bool_lines = []
+        other_lines = []
+        IS_BOOL_REGEX = re.compile(r'\s*bool ')
+        for lines in contents_lines:
+            for line in lines.split('\n'):
+                if IS_BOOL_REGEX.match(line):
+                    bool_lines.append(line)
+                else:
+                    other_lines.append(line)
+        result += '\n'.join(other_lines)
+        result += '\n'
+        result += '\n'.join(bool_lines)
 
         if Globals.protoc_insertion_points:
             result += '\n/* @@protoc_insertion_point(struct:%s) */' % self.name
```

PetteriAimonen, 2022-02-20:
Easiest would be to place the `has_x` fields at the beginning of the struct. Then it would be possible to just add a flag to `pb_msgdesc_t` that tells whether they are in the beginning or next to each field. Handling that flag in `pb_common.c` would be quite simple.

Ideally the `has_` fields would be defined as bitfields, to minimize the storage required. That has some portability problems, but if supported on the platform it would give significant savings.

PetteriAimonen, 2022-05-30:
Currently there are no plans to implement this.

closed 2022-05-30 by PetteriAimonen