nanopb/nanopb #757: When Using google.protobuf.Any within a message, -DPB_FIELD_32BIT does not set field size to 32bits

mstiehm-NF, 2022-02-22:
I am creating an envelope message like the example in tests.
In the any.options file I have a large max_size set for value.
I compile with -DPB_FIELD_32BIT
In_ my envelope.options file I have set my payload (Any) max_size to the same as the any.options file, but during compilation, I get the assert described in #680.

Upon inspection I have found that my envelope.pb.c still has 2 set for width.

Is there something else I should be doing, or is this not properly setting the width?
(I also have other proto files with repeated bytes that are being set correctly to 32 bit)

PetteriAimonen, 2022-02-22:
Do you have a .proto file that reproduces this problem?

As a workaround you can try `* descriptorsize = DS_4` in options file.

PetteriAimonen, 2022-02-22:
You can also decode the envelope manually with `pb_decode_tag()`, like in `examples/using_union_messages`.

If you are trying to allocate a huge struct on the stack, you are probably overflowing your allocated stack space which could explain the segfault.

mstiehm-NF, 2022-02-22:
That could be. I'm trying to make the max size 8mb.

The majority of the sub-messages I'm encoding are only around 100-150 bytes, but I am encoding images that could be up to 8mb. That's why I think callbacks might work the best here since I would only allocate the memory when I need it.

mstiehm-NF, 2022-02-23:
Update: It looks like it could be how I'm building the files. Does this look like it should work?
```# Makefile adapted from nanopb/examples/using_union_messages
# Include the nanopb provided Makefile rules
include ../nanopb-0.4.5-linux-x86/extra/nanopb.mk

# Compiler flags to enable all warnings & debug info
CFLAGS = -Wall -Werror -g -DPB_FIELD_32BIT
CFLAGS += -I$(NANOPB_DIR)

proto:
	../nanopb-0.4.5-linux-x86/generator-bin/nanopb_generator ../interface/schema/google/protobuf/*.proto -I ../interface/schema/google/protobuf/ -D google/protobuf/
	../nanopb-0.4.5-linux-x86/generator-bin/nanopb_generator ../interface/schema/*.proto -I ../interface/schema/

all: test_client test_server

.SUFFIXES:

clean:
	rm -f test_client test_server *.pb.c *.pb.h google/protobuf/*.pb.c google/protobuf/*.pb.h

%: %.c google/protobuf/*.pb.c *.pb.c common.c data/medMap.c
	$(CC) $(CFLAGS) -o $@ $^ $(NANOPB_CORE)```

The dirs are like this
repo
    nanopb
    interface
        schema
    tests
        Makefile
So, I'm building the files into the tests folder

PetteriAimonen, 2022-02-23:
I don't see anything special there.

But do note that e.g. on Linux, default stack size for 32-bit is 2 MB and for 64-bit it is 8 MB.
On embedded platforms or in kernel mode of course much smaller.
So *if* you are allocating a 8 MB struct on stack, it is no surprise it is crashing.
Make it static or smaller.

PetteriAimonen, 2022-02-23:
For repeated fields, you'll want to set `max_count` also. Size is the size of each bytes entry, count is the number of entries in the array.

mstiehm-NF, 2022-02-23:
Ah, that makes sense. When using repeated bytes, does it pack a tag for each byte If I were to use max_count:512000 max_size:1? Or would it pack the same as using max_count:1 max_size:512000?

PetteriAimonen, 2022-02-23:
Each array entry will get a tag and length byte.

mstiehm-NF, 2022-02-23:
Okay. Thanks for the help. Everything is now working using the `descriptorsize: DS_4`, but without it I still get the issue.
Here is the offending proto file.
```
syntax = "proto3";

import "google/protobuf/any.proto";
import "time.proto";

package ui;

//
// Common Envelope of data that is sent.
// For Every request there will always be an envelope returned
//
message Envelope
{
  Time timestamp = 1;    // Time that the message was created (This is not meant to be in the same time frame of both computers)
  uint32 sequence = 2;   // Incrementing number to keep track of message order
  PayloadType type = 3;  // The message type in the any object
  google.protobuf.Any payload = 4;
}
```

PayloadType is an Enum.

ShaharHD, 2022-02-28:
I'm having the exact same issue, even with simple protobuf protocol:

```protobuf
// Calibration data maps a raw load cell value to a weight value in grams
message Calibration {
  reserved 1, 2;

  repeated sint32 loadcell_array = 3; // Array of raw loadcell values
  repeated sint32 weight_array = 4; // Arrays of actual weight in grams
}
```

options file:

```protobuf
// * descriptorsize: DS_AUTO

// Max number of calibration points
Calibration.*_array max_count:10
Calibration.weight_array int_size:IS_16
```

generated code looks fine

```c
typedef struct _Calibration {
    pb_size_t loadcell_array_count;
    int32_t   loadcell_array[10]; /* Array of raw loadcell values */
    pb_size_t weight_array_count;
    int16_t   weight_array[10]; /* Arrays of actual weight in grams */
} HyGearBLEProtocol_Calibration;
```

Tried with both changing `* descriptorsize: DS_AUTO` as suggested in #757 and also using `PB_FIELD_32BIT` as suggested in #680

Using latest nanopb (0.4.5) on MacOS M1 Max running latest monterey.

PetteriAimonen, 2022-05-30:
This is probably fixed by #720.

closed 2022-05-30 by PetteriAimonen