nanopb/nanopb #759: Option to skip imports?

jaskij, 2022-03-01:
In one of my proto files, I am importing `google.protobuf.Timestamp`, which is not used by the code using nanopb (it's filled in later, the MCU does not have wall clock time). My problem is that the generated `pb.h` file still includes `google/protobuf/timestamp.pb.h` - while sure, I can generate and include it in the project, this feels unnecessary. Manually editing generated files is of course a Bad Idea (tm), which I'm reluctant to do.

Is there a way to skip generating the `#include` referencing a specific import?

PetteriAimonen, 2022-03-01:
There is command line switch `-x FILE` `              Exclude file from generated #include list.` for nanopb_generator.py.

jaskij, 2022-03-01:
Ah, thanks, I looked through the API docs thinking perhaps it will be available as an option.

closed 2022-03-01 by jaskij

jaskij, 2022-03-02:
I went off the docs at https://jpa.kapsi.fi/nanopb/docs/reference.html - are those out of date, or is this simply not in a release version yet?

(I also only searched for two words `import` and `skip` which would have missed it regardless)

PetteriAimonen, 2022-03-02:
Not really out of date, it's just condensed. *"The generator behaviour can be adjusted using several options, defined in the [nanopb.proto](https://github.com/nanopb/nanopb/blob/master/generator/proto/nanopb.proto) file in the generator folder. Here is a list of the most common options, but see the file for a full list."*

The list of rare generator options not needed by many people has grown over the years.