nanopb/nanopb #761: Compilation fails for code generated with ccarm compiler

kkarnatak, 2022-03-03:
I am trying a modified version of examples/network_server on ARM platform. The code works perfectly when I compile it using gcc on ubuntu. However, when I use armcc compiler, I get the following errors:

1. All the lines where the macros generated from the proto files are assigned:

    "network_test.c", line 222: error #29: expected an expression
            NwContainer message_dec = NwContainer_init_default; 

2. In the common.c file of network_server example, where data is read or written over the socket.

     common.c", line 41: error #28: expression must have a constant value        
           pb_istream_t stream = {&read_callback, (void*)(intptr_t)fd, SIZE_MAX};

I suspect it has to do with the braces {} but I was wondering if there is any option to generate code using nanopb especially ARM platforms? If not, how to resolve these issues?

PetteriAimonen, 2022-03-03:
The second one should be quite simple to work around in your own code, if armcc just cannot put a function pointer inside brace initializer.

The first one sounds like a bug in armcc. You can try using `= {}` to initialize to zeros, `pb_decode()` sets the defaults anyway.

PetteriAimonen, 2022-03-03:
If the compiler has a selection for standard version, it's worth trying e.g. C11. It might default to 1989 version or something.

kkarnatak, 2022-03-30:
On C11 everything works without issues. But C99 requires the changes which were discussed above, following which everything works as expected. Hence, closing the issue. Thanks for the input and help!