nanopb/nanopb #762: Generator/protoc script can end up invoking itself

PetteriAimonen, 2022-03-08:
If the `grpcio-tools` Python package is not available, `generator/protoc` script attempts to run `protoc` from system `PATH`.
If the `nanopb/generator` directory is prepended to system `PATH`, it ends up invoking itself in an endless loop.

jaskij, 2022-03-08:
Thinking about this, I believe if `generator/protoc` *is* the first in `PATH`, it should either:

- refuse to work (since if it's in `PATH`, odds are the original `protoc` can find the nanopb plugin anyway), but this requires somehow removing `generator/protoc` from the distributed files
- iterate over `PATH` looking for Google's `protoc` (how will it find that?)

jaskij, 2022-03-09:
That certainly *is* the simplest way, I do wonder if there would be side effects (eg for spawning subprocesses or assumed paths elsewhere in the script). But it shouldn't be the case, looking how using `protoc --plugin` works.

In my build container I have resorted to just removing `generator/protoc` before adding `generator` to `PATH`. This works nicely since `protoc` automatically looks for plugins in `PATH`. I can then simply call `protoc` without any extra flags to pass the plugin path.

PetteriAimonen, 2022-03-09:
Yeah.. the main idea of adding the `protoc` wrapper script was to avoid having to modify PATH in the first place, i.e. calling `..../path/to/nanopb/generator/protoc` would have the correct plugin path set up automatically.

closed 2022-05-30 by PetteriAimonen