nanopb/nanopb #763: How to avoid creating new has_<field_name> with proto3?

px528, 2022-03-09:
when we define a message like
message test
{
    repeated uint32 data=1;
}

the c structure will contain a new field "has_data", 
as "data" is regarded as optional field in proto3.

do we have a way to not to create this new field?

PetteriAimonen, 2022-03-09:
Hmm, that should really be giving `data_count` because it is a repeated field and not optional.
At least unless you have some weird settings in `.options` file.

px528, 2022-03-09:
Your E-mail has been received and I will reply as soon as possible

px528, 2022-03-09:
yeah,haha,do we have such option?

PetteriAimonen, 2022-05-01:
I have trouble understanding your writing, but you can see the list of available options [here](https://github.com/nanopb/nanopb/blob/master/generator/proto/nanopb.proto).

closed 2022-05-01 by PetteriAimonen

ABD-01, 2024-08-23:
Hi!
Is there any solution for this yet??
I am using v0.4.8
And for each nested nested message, there is has_xxx field generated. 
I too want to avoid it.

PetteriAimonen, 2024-08-23:
@ABD-01 Maybe the solution in #962 helps?

ABD-01, 2024-08-23:
Ahh Thanks a lot
I was a bit impatient.
I updated `proto3_singular_msgs` to `true` in the file you mentioned above
https://github.com/nanopb/nanopb/blob/e34dfae6f1aeb87cbe1c05a8fb96f5b9cb1034f6/generator/proto/nanopb.proto#L95

build nanopb again using the [tools/make_windows_package.sh](https://github.com/nanopb/nanopb/blob/master/tools/make_windows_package.sh)

Worked 👍

ABD-01, 2024-08-23:
Also found out that this is enough actually.
```
nanopb_generator.exe hello.proto -s proto3_singular_msgs:true
```