nanopb/nanopb #766: armcc compiler warning with generated *.pb.c files

shahsahi, 2022-03-17:
I am running into a warning when compiling the *.pb.c files generated from nanopb, as follows:

```
streams.pb.c:54:1: warning: variable length array folded to constant array as an extension [-Wgnu-folding-constant]
PB_BIND(MetaData, MetaData, AUTO)
pb.h:481:27: note: expanded from macro 'PB_BIND'
    msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
                          ^
<scratch space>:21:1: note: expanded from here
PB_GEN_FIELD_INFO_ASSERT_AUTO
```

It seems this comes from the PB_BIND macro and how const ints are treated in C when used as an array size. Any guidance on if/how this can be resolved?

PetteriAimonen, 2022-03-17:
Try setting the compiler to C11 standard mode.

PetteriAimonen, 2022-03-17:
@shahsahi Could you test one thing, it would help me fix this by default in nanopb code. Add this code to one of your '.c' files and tell me if it gives any warnings when compiled **without C11 mode**:

```
#ifndef __ARMCC_VERSION
#warning MISSING_ARMCC_VERSION
#endif
#if defined(__STDC_VERSION__) && __STDC_VERSION__ >= 201112L
#warning C11 mode active
#endif
_Static_assert(1, "STATIC_ASSERT_TEST");
```

shahsahi, 2022-03-17:
Yes, I did this and got only the "MISSING_ARMCC_VERSION" warning. I think I mistakenly said armcc compiler in my title - `__GNUC__` is defined.