nanopb/nanopb #767: example for interleaved array with fixed count

px528, 2022-03-22:
Hello,

recently I met a decoding issue as below notes mentioned.
`Note: The decoder only keeps track of one fixed_count repeated field at a time. Usually this it not an issue because all elements of a repeated field occur end-to-end. Interleaved array elements of several fixed_count repeated fields would be a valid protobuf message, but would get rejected by nanopb decoder with error "wrong size for fixed count field"`

I just want to try to make things clear,
does this mean we can't support a array like uint8 a[3][4]?
or we can't support below struct?
`typedef strcut
{
uint32 a[4];
uint32 b[2];
}test;`

in my examples are not good, can you help provide some example that we can't support?

PetteriAimonen, 2022-03-22:
It's related to the protobuf wire encoding format.
Every decoder that I know of will emit the fields in order:

     a[0] a[1] a[2] a[3] b[0] b[1]

But this is valid protobuf format also:

    a[0] b[0] a[1] a[2] b[1] a[3]

Because the decoder only tracks one fixed_count array at a time, this message would be rejected.

In practice this could occur if messages are merged by concatenating their encoded representations.

closed 2022-03-23 by px528

px528, 2022-10-11:
Your E-mail has been received and I will reply as soon as possible