nanopb/nanopb #768: Pass arguments to protoc

a1lu, 2022-03-29:
Hi,
I use nanopb within my cmake project and need to pass
`--experimental_allow_proto3_optional` to protoc
I use nanopb 0.4.5 with the bundled protoc 3.13.0.

The test case proto3_optional adds the flag as a env variable in the SConscript file. But this does not work for me. 

What is the correct way to pass the flag to protoc?
I added a ${nanopb_protoc_flags} variable to the custom command in FindNanopb.cmake:276.
I can create a pull request later if you like this approach.

This is maybe related to #628
Thanks.

PetteriAimonen, 2022-03-29:
Yes, having a variable to pass options to protoc in CMake makes sense.

Google's C++ protobuf cmake rules take `PROTOC_OPTIONS`. That's probably a reasonable name to use here also.
I added it now.

a1lu, 2022-03-29:
Thanks, this fixed it for me.

closed 2022-03-29 by a1lu