nanopb/nanopb #770: Optimising size of enum

kkarnatak, 2022-03-30:
I have enum in my protobuf file. All the values can be represented in 1 byte. Currently, I am compiling source code using gcc on ubuntu x64 platform and restricted to C99 spec. Final target platform is ARM. 

The size of enum is 4 bytes, leaving 3 bytes wasted. 
As far as I know, as opposed to C++, in C we cant specify type for the enum. I am curious if there is a way to optimise this? 

///////////////////////
SNIPPET FROM PROTO FILE
///////////////////////

enum OptionalField{
	ID_0 = 0;
	ID_1 = 1;
	ID_2 = 2;
	ID_3 = 3;
	ID_4 = 4;
	ID_5 = 5;
	ID_6 = 6;
	ID_7 = 7;
}

OptionalField optionalField = 1;

/////////////////////////////////////////

GENERATED CODE FROM NANOPB

/////////////////////////////////////////

/* Enum definitions */
typedef enum _MainMenu_OptionField { 
    MainMenu_OptionField_ID_0 = 0, 
    MainMenu_OptionField_ID_1 = 1, 
    MainMenu_OptionField_ID_2 = 2, 
    MainMenu_OptionField_ID_3 = 3, 
    MainMenu_OptionField_ID_4 = 4, 
    MainMenu_OptionField_ID_5 = 5, 
    MainMenu_OptionField_ID_6 = 6, 
    MainMenu_OptionField_ID_7 = 7 
} MainMenu_OptionField;

Size of enum MainMenu_OptionField is : 4 bytes

PetteriAimonen, 2022-03-30:
You are probably looking for option `packed_enum = true`.
See e.g. https://github.com/nanopb/nanopb/blob/master/tests/enum_sizes/enumsizes.proto

kkarnatak, 2022-03-30:
Thanks for the prompt reply. I added it and pb generated __attribute__(packed)) along with the structure.

closed 2022-03-30 by kkarnatak