nanopb/nanopb #772: new (global) option like "FT_DEFAULT" but fallback to FT_POINTER instead of FT_CALLBACK

ConradWood, 2022-04-09:
Hi,

I propose to introduce at least one extra option to add specific functionality.
The functionality I like to introduce is to generate fields as static fields, if possible, but fields with variable length to be allocated dynamically via malloc (realloc).
Currently, the behaviour of FT_DEFAULT is to generate static fields if it can and generate callbacks if it cannot create static fields.
I propose to keep FT_DEFAULT behaviour "as-is" for now.

Instead, I propose to create a new option, called "FT_AUTO_FB_POINTER". 
For example: "nanopb_generator.py -s type=FT_AUTO_FB_POINTER [src]".

The behaviour of the options shall be identical to FT_DEFAULT, except, it would use pointers and malloc to generate dynamic fields.

in "nanopb_generator.py" I see:

  if field_options.type == nanopb_pb2.FT_DEFAULT:
            if can_be_static:
                field_options.type = nanopb_pb2.FT_STATIC
            else:
                field_options.type = nanopb_pb2.FT_CALLBACK

so IMHO the change is fairly trivial. We would *add* below:

  if field_options.type == nanopb_pb2.FT_AUTO_FB_POINTER:
            if can_be_static:
                field_options.type = nanopb_pb2.FT_STATIC
            else:
                field_options.type = nanopb_pb2.FT_POINTER

(of course, also modifying the options.proto files for enums).

Note that the naming scheme "FT_AUTO_FB_POINTER" is intended to be extensible.
Perhaps someone, some day, wants to add FT_AUTO_FB_POINTER_FOR_STRING_AND_CALLBACK_FOR_ARRAY
(the example might be a little long, but it's for illustrative purposes only :) )
(FB stands for "Fallback")

Would a patch to that extent be welcome?

Conrad

PetteriAimonen, 2022-04-09:
I think it might be better to have a separate `optional FieldType fallback_type = x [default = FT_CALLBACK]` option in `NanoPBOptions`. This would allow fallback to `FT_IGNORE` also.

With regards to specific field types based on datatype, I think we could eventually have more flexible selectors in `.options` file. Something like `[datatype:TYPE_BYTES] * type:FT_POINTER`

ConradWood, 2022-04-09:
On Sat, 2022-04-09 at 11:17 -0700, Petteri Aimonen wrote:
> I think it might be better to have a separate optional FieldType
> fallback_type = x [default = FT_CALLBACK] option in NanoPBOptions.
> This would allow fallback to FT_IGNORE also.
> With regards to [...]

That makes a lot more sense to me as well.
Would the corresponding "-s" option be "FALLBACK_TYPE"?
e.g. 
-s "FALLBACK_TYPE:FT_POINTER" ?

Conrad

PetteriAimonen, 2022-04-10:
It would be `-s fallback_type:FT_POINTER`, I think. I think protobuf text format parsing (which `-s` uses) is case-sensitive.

If this is something you would like to work on, a pull request is very welcome!

ConradWood, 2022-04-10:
pull request is here[1].
Kindly requesting your attention and review.

[1]
https://github.com/nanopb/nanopb/pull/773

closed 2022-05-30 by PetteriAimonen