nanopb/nanopb #775: custom 'realloc' with a context

ConradWood, 2022-04-21:
Hi,

In our firmware is a concept of 'criticality' of different sections. The firmware offers functionality via an API and manages partitions for different binary sections to execute. So, on each module might be 2 or more different sections running. For example, a 'critical', real-time machine controlling section and a 'user display', not-quite-as-critical section.
Any section can call the firmware with a pre-assigned handle. The kernel then prioritises resources depending on what's critical and what is not. 
For nanopb, we'd like to use different 'realloc' strategies and ram areas, depending on the section calling "pb_decode()". 
I'm pondering how to do that.
My current thinking is to add a new function 
bool pb_decode_ex2(pb_istream_t *stream, const pb_msgdesc_t *fields, void *dest_struct, unsigned int flags,void *opaque_context);

Then I would pass the opaque_context to pb_decode_inner and along the functions all the way to custom_realloc().

I would then define pb_realloc as:
#define pb_realloc(ptr, size,opaque_context) custom_realloc(ptr, size,opaque_context)

the default definition would change to:
#define pb_realloc(ptr, size,opaque_context) realloc(ptr, size)

The API would remain backwards compatible. 
This proposal is is also the least intrusive way I can come up with OTH.
It would only deployments of nanopb if a) PB_ENABLE_MALLOC is true AND b) a custom realloc is supplied

Stack usage might increase by a few bytes to push the extra pointer on the stack.

I'm happy to do a patch if that patch is likely to be accepted.

Conrad

PetteriAimonen, 2022-04-21:
I agree with your use case.

My thoughts would be:

1. `pb_istream_t` is already used as a kind of context for decoding, as it stores also the error status. It is already available in most functions calling `reallocate_field()`. Notably it is not available in `pb_release()`.
2. Changing `pb_realloc` definition would break existing applications that redefine it. An alternative would be to define a new `pb_realloc_ctx()` that by default calls `pb_realloc()`.

On the other hand, without changing nanopb code, you could use thread-local variables to pass information to the custom realloc implementation.

ConradWood, 2022-04-22:
On Wed, 2022-04-20 at 22:40 -0700, Petteri Aimonen wrote:
> I agree with your use case.
> My thoughts would be:
>    1. pb_istream_t is already used as a kind of context for decoding,
> as it stores also the error status. It is already available in most
> functions calling reallocate_field(). Notably it is not available in
> pb_release().

-v please :) 
Are you suggesting to add a field to pb_istream_t, for example 
"void *context" and pass that to realloc? or pass the pb_istream_t to
realloc?

>    2. Changing pb_realloc definition would break existing
> applications that redefine it. An alternative would be to define a
> new pb_realloc_ctx() that by default calls pb_realloc().

ah yes. makes sense to define a pb_realloc_ctx.

> On the other hand, without changing nanopb code, you could use
> thread-local variables to pass information to the custom realloc
> implementation.

Good suggestion. It would need to make sure that pb_decode() is not
invoked by an IRQ though.
It does feel a bit like a side-channel to the API though, that is, it's
a bit harder to argue over code that works like that.

Conrad

PetteriAimonen, 2022-04-22:
I would suggest passing `pb_istream_t` to the realloc function. Currently it has a single `state` pointer inside it. When #776 is done, one could also subclass / struct composite the `pb_istream_t` to a custom structure that contains as much state as you need.

But now that I think of it, it might even make sense to have a define for `PB_ENABLE_MALLOC_CONTEXT`. This could then have both `realloc()` and `free()` functions as pointers in the `pb_istream_t`. That would allow easily use completely unrelated implementations in different contexts.

ConradWood, 2022-04-24:
On Fri, 2022-04-22 at 05:49 -0700, Petteri Aimonen wrote:
> I would suggest passing pb_istream_t to the realloc function.
> Currently it has a single state pointer inside it. When #776 is done,
> one could also subclass / struct composite the pb_istream_t to a
> custom structure that contains as much state as you need.

> But now that I think of it, it might even make sense to have a define
> for PB_ENABLE_MALLOC_CONTEXT. This could then have both realloc() and
> free() functions as pointers in the pb_istream_t. That would allow
> easily use completely unrelated implementations in different
> contexts.

Yes, I think that's another good option.
I'm thinking of something like:

struct pb_istream_s
{
...
#ifdef PB_ENABLE_MALLOC_CALLBACKS
void *(*realloc)(pb_istream_t *stream,void *ptr, size_t size);
void (*free)(pb_istream_t *stream,void *ptr);
#endif
}

PetteriAimonen, 2022-04-24:
Looks reasonable.

Potentially one could have just `*realloc`, as most platforms specify it as *"if size is equal to zero, and ptr  is  not  NULL,  then  the  call  is  equivalent  to free(ptr)"*. Not sure if is would simplify implemention in user code, or if it is just confusing.

ConradWood, 2022-04-25:
On Sun, 2022-04-24 at 10:15 -0700, Petteri Aimonen wrote:
> Looks reasonable.
> Potentially one could have just *realloc, as most platforms specify
> it as "if size is equal to zero, and ptr is not NULL, then the call
> is equivalent to free(ptr)". Not sure if is would simplify
> implemention in user code, or if it is just confusing.
> —

I thought it's non obvious, it does save some bytes. I would drop the
pointer to free() and put a a comment to that extent in the code, so it
becomes more obvious.
I am happy to proceed and write a few testcases as well.

Conrad

PetteriAimonen, 2022-04-25:
Sounds good to me!

PetteriAimonen, 2022-09-14:
Looks like C17/C23 deprecates the `realloc()` with zero size.

PetteriAimonen, 2024-09-19:
It would make sense to figure something out for this use case by the 1.0 release, but I'm not quite satisfied with the current implementation.

It might make sense to have a context structure apart from the stream structure.

PetteriAimonen, 2024-09-20:
Basic idea sounds good, though I'm a bit worried about the stack size increase.
The default context has to be stack allocated to support multithreading.

On the other hand, if #776 is done, then the stream could be the context, avoiding the extra structure.
Architecturally it's not as clean, but it has benefits for custom streams too.

Yet another aspect is that #571 would need some kind of custom allocation logic that would avoid realloc() when possible. I'm not sure if that will require some kind of context.

ConradWood, 2024-09-20:
I might need to digger deeper into the code again, I haven't kept up to
date with latest development, but - perhaps a bit OTOH - my impression
is that

A default context could be static in flash. The context could be as 
simple as:

typedef struct pb_context {
 void *opaque; // not used by pb itself, NULL in default
}

(or maybe even simpler: typedef void *; )

This assumes we link against custom malloc/free functions defined as
 void *pb_malloc (struct pb_context *ctx, size_t size);
 void *pb_free (struct pb_context *ctx, void *adr);

instead of malloc() and free().

it is then upto the user to provide wrappers around malloc() and
free(). The linker will throw a helpful warning during migrations.
it does require a few extra bytes of code for the wrappers, but with
inline and no_instrumentation that could be quite minimal

as per #571 - I think the realloc dependency is a good call to remove,
I'm not a big fan of realloc, it can indeed be a bit tricky. It does
seem to me like an unrelated issue though. Maybe a candidate for a
compile option instead? 

as per #776 - it is definitely useful to store extra_state. I did some
hacks in the past to work around that and would benefit from an
extra_state in stream. As per using it as context, I wonder how this
would affect, for example, pb_release(). I assume pb_release eventually
calls free(). free() would need the same context used in malloc() to
figure out which memoryarea to use. Are you thinking of changing
pb_release to include the stream? and pass the stream to free() and
malloc()?

Conrad

PetteriAimonen, 2024-09-20:
Hmm, I agree that the default context could be ROM at least for now.

Yeah, I'm considering those possibilities.

ConradWood, 2026-02-02:
> PetteriAimonen
> @ConradWood [2] If you have a moment of time, can you share your
> thoughts on this:
> Now pb_decode_ctx_t has a pointer to pb_allocator_t. Idea being that
> same allocator can be shared between multiple decoding contexts.
> Example code [3]
> But should pb_release() take a pb_decode_ctx_t or a pb_allocator_t?
> It only needs the allocator, but on the other hand it kinda feels
> better parallel to pb_decode() if it takes the decode context. But
> then again the decode context may no longer be relevant by the point
> when you are releasing the message.
> Currently dev_1.0 branch has bool pb_release(pb_decode_ctx_t *ctx,
> ... but I'm leaning towards changing it to bool
> pb_release_s(pb_allocator_t *ctx, ....

will do. I will need a few days to digest all the changes though.

[1] (nanopb/nanopb#775)
    https://github.com/nanopb/nanopb/issues/775#issuecomment-3831203373
[2] @ConradWood https://github.com/ConradWood
[3] Example code
    https://github.com/nanopb/nanopb/blob/dev_1.0/tests/custom_allocator/decode_alltypes_allocator.c#L200-L203
[4] unsubscribe
    https://github.com/

ConradWood, 2026-02-18:
> @ConradWood [3] If you have a moment of time, can you share your

Sorry it took so long - my thoughts are as below:

1) I would (slightly) prefer to pass the ctx to realloc() and to free() instead of the allocator itself. this is because
a) "it kinda feels better parallel to pb_decode()" ;-)
b) we got a link from ctx to allocator (ctx->allocator) but not vice versa. This means we'd end up tracking it in some sort 
'state' in the allocator. This then means we need one allocater per stream. And finally that negates the (excellent) idea
of reusability of the allocator.

As per your point that the "decode context may no longer be relevant by the point when you are releasing the message.".
I assumed the api contract to be such, that once the context is 'released' (no longer be relevant) the allocator is no longer relevant
any more either. That is, any references to allocated memory must have been released prior to the context being released.
Is that not the case?

2) I was pondering about the 'state' variable in the allocator for a bit. It wasn't immediately clear to me that it is
never read, nor written but preserved by nanopb. Is that the future api guarantee as well?
if so, let's say this clearly in the comment: 
"this field is neither read from nor written to by nanopb, but preserved as-is."

3) I rendered documentation with doxygen (https://www.doxygen.nl/). [1] I found that very useful.
Would you include its configfile and run doxygen Doxyfile during build? 
The config is here:
https://www.singingcat.net/static/nanopb/Doxyfile
This is the output doxygen created for nanopb:
https://www.singingcat.net/static/nanopb/doxygen-output/html/annotated.html

Saying all that: Even as-is proposed, the ctx/allocator mechanism will help with our usecase (YAY!!! thank you so much!). 

Conrad

[1] https://www.doxygen.nl/). https://www.doxygen.nl/
[2] (nanopb/nanopb#775)
    https://github.com/nanopb/nanopb/issues/775#issuecomment-3831203373
[3] @ConradWood https://github.com/ConradWood
[4] Example code
    https://github.com/nanopb/nanopb/blob/dev_1.0/tests/custom_allocator/decode_alltypes_allocator.c#L200-L203
[5] unsubscribe
    https://github.com/