nanopb/nanopb #778: .options not taken into account if CWD is not root of protocol directory tree

jaskij, 2022-05-04:
I have a directory tree with my protocol files, which I tried to use with [buf.build](https://buf.build/). Unfortunately, `.options` files were not working - seemingly because the protocol files were in a subdirectory. All my `.options` files are besides matching `.proto` files (ie `data.options` in the same directory as `data.proto`).

Consider this (with `protoc-gen-nanopb` in path):

```shell
$ protoc -Iproto --nanopb_out=nanopb_out $(find . -type f -name '*\.proto') # generates wrong files, disregarding .options
$ cd proto
$ protoc -I. --nanopb_out=nanopb_out $(find . -type f -name '*\.proto')  # works as expected
```

I believe this behavior to be a bug, as the `.options` files should be taken into account in both cases.

PetteriAimonen, 2022-05-04:
Unfortunately `protoc` doesn't pass the full path of the input file to plugins and changing that behavior turned out to be quite difficult (https://github.com/protocolbuffers/protobuf/pull/811)

Alternatives:

- Call `nanopb_generator.py -Iproto proto/foo.proto`
- Pass `--nanopb_opt=-Iproto` to give the include path to the plugin

jaskij, 2022-05-04:
>     * Call `nanopb_generator.py -Iproto proto/foo.proto`

This worked perfectly fine, thank you.

closed 2022-05-30 by PetteriAimonen