nanopb/nanopb #781: Support repeated Any Message

AravindGopala, 2022-05-23:
Is there a way to encode a repeated google.protobuf.any message using nano pb?

message RepeatedAny{
repeated google.protobuf.any sensors = 1;
}

PetteriAimonen, 2022-05-23:
Sure, it is just a regular message.

https://github.com/nanopb/nanopb/tree/master/tests/any_type shows how to encode a single Any message, encoding many is like encoding any array. You'll have a choice between allocating statically, allocating dynamically or using callbacks. Or you can just encode a single subfield at a time into output stream, because concatenating encoded concatenates arrays in protobuf format.

PetteriAimonen, 2022-05-23:
Also, please link posts together if you feel the need to post the same question immediately to many places: https://stackoverflow.com/questions/72342189/how-to-encode-a-repeated-google-protobuf-any

Not doing so just wastes people's time when same question ends up answered multiple times.

closed 2022-05-23 by PetteriAimonen

AravindGopala, 2022-05-23:
Hi Petteri,  I think I found my issue, I cannot use:
message RepeatedAny{
repeated google.protobuf.any sensors = 1;
}
```

Instead I should use something like this
```
message SensorAny{
google.protobuf.any sensor = 1;
}

message RepeatedAny{
repeated SensorAny sensors = 1;
}
```

I should not use the repeated tag on the google.protobuf.any, I should be using it on a message that contains the google.protobuf.any, so that the protobinary can contains `<RepeatedAny><RepeatedAny>......`

Below is the sample code, if someone finds this question in the future for nanopb:
```

    /* First encode the Sensor Any message as the value of the SensorAny,
       The first filed of this message is of type google.protobuf.any, so it should have
       1. message.type_url
       2. message.vale
    */

    void* pBufAny = calloc(1, sBufSize);
    pb_ostream_t ostream_any = pb_ostream_from_buffer(pBufAny, sBufSize);
    SensorAny SensorAnyProto = SensorAny_init_default;
    SensorAnyProto.has_message = true;
    SensorAnyProto.message.type_url.arg = "type.googleapis.com/SensorAny.proto";
    SensorAnyProto.message.type_url.funcs.encode = Proto_encode_string;
    ProtoEncodeBufferInfo_t BufInfo = {
            .Buffer = pBuf,
            .BufferSize = ostream.bytes_written,
    };
    SensorAnyProto.message.value.funcs.encode = Proto_encode_buffer;
    SensorAnyProto.message.value.arg = &BufInfo;
    pb_encode(&ostream_any, proto_platform_services_fileingestion_FisSubmission_fields, &SensorAnyProto);
    free(pBuf);

    // Now Use the above encoded Any message buffer pBufAny to set the first repeated field in RepeatedAny

    RepeatedAny SensorAnyRepeated = RepeatedAny_init_default;
    ProtoEncodeBufferInfo_t AnyBufInfo = {
            .Buffer = pBufAny,
            .BufferSize = ostream_any.bytes_written,
    };

    AnyRepeated.sensors.arg=&AnyBufInfo;
    AnyRepeated.sensors.funcs.encode = Proto_encode_buffer;

    void* pBufAnyRepeated = calloc(1, sBufSize);
    pb_ostream_t ostream_repeated = pb_ostream_from_buffer(pBufAnyRepeated, sBufSize);
    !pb_encode(&ostream_repeated, RepeatedAny_fields, &AnyRepeated);
    free(pBufAny);
```
I will update the stack overflow thread as well, thanks: 
```