nanopb/nanopb #782: Allow FT_POINTER without PB_ENABLE_MALLOC

Jonathan-Povirk-MSASafety, 2022-05-23:
**What should the feature do?**

Currently, attempting to use an `FT_POINTER` type without `PB_ENABLE_MALLOC` will fail `PB_RETURN_ERROR(..., "no malloc support");`.
This proposal would allow users to statically allocate `FT_POINTER` types, decoupling the storage of e.g. a `bytes` type
from the generated protobuf struct.

**In what situation would the feature be useful?**

Our company uses a common protobuf repo that defines a top-level CommonMessage type and some common vocabulary types.
Because of the number of build systems we support, this common protobuf repo is distributed as generated C code.
Each project defines its own unique protos that can be serialized into this CommonMessage.
Because of this, we cannot know the maximum static size when we compile the common protobuf repo.

The a simplified version of the CommonMessage protobuf resembles the following:

```proto
message CommonMessage {
  uint64 time        = 1;
  uint32 message_id  = 2;
  bytes message_data = 3;
}
```

with `FT_POINTER` set for `message_data`.

The following C snippet shows what `FT_POINTER` usage without malloc may look like:

```c
int main(void)
{
  PB_BYTES_ARRAY_T(SomeOtherProtobuf_size) message_data;

  // Create SomeOtherProtobuf and encode it
  SomeOtherProtobuf other = CreateSomeOtherProtobuf();
  message_data.size = EncodeSomeOtherProtobuf(message_data.bytes, &other);

  CommonMessage msg;
  msg.time = TimeNow();
  msg.message_id = SomeOtherProtobuf_msgid;
  msg.message_data = (pb_bytes_array_t*)&message_data;

  pb_byte_t encoded[CommonMessage_size + SomeOtherProtobuf_size];
  pb_ostream_t os = pb_ostream_from_buffer(encoded, sizeof(encoded));
  assert(pb_encode(&os, ctl_ConnectivityTransaction_fields, &msg));

  // Reset message and message_data
  memset(&msg, 0, sizeof(msg));
  memset(message_data.bytes, 0, sizeof(message_data.bytes));

  // Set the size of message_data in msg
  message_data.size = sizeof(message_data.bytes);
  msg.message_data = (pb_bytes_array_t*)&message_data;

  // message_data->size is used to decode
  pb_istream_t is = pb_istream_from_buffer(serialized, os.bytes_written);
  assert(pb_decode(&is, ctl_ConnectivityTransaction_fields, &msg));
```

This would also allow multiple large protobuf structs to potentially share the same `pb_bytes_array_t`, saving memory
in statically allocated contexts.

This scenario appears to be possible given this very naive patch at `c912413`.

```patch
diff --git a/pb_decode.c b/pb_decode.c
index b194825..d5d936b 100644
--- a/pb_decode.c
+++ b/pb_decode.c
@@ -621,9 +621,16 @@ static void initialize_pointer_field(void *pItem, pb_field_iter_t *field)
 static bool checkreturn decode_pointer_field(pb_istream_t *stream, pb_wire_type_t wire_type, pb_field_iter_t *field)
 {
 #ifndef PB_ENABLE_MALLOC
+  if ((*(pb_size_t*)field->pField) != 0)
+  {
+    return decode_static_field(stream, wire_type, field);
+  }
+  else
+  {
     PB_UNUSED(wire_type);
     PB_UNUSED(field);
     PB_RETURN_ERROR(stream, "no malloc support");
+  }
 #else
     switch (PB_HTYPE(field->type))
     {
@@ -918,6 +925,7 @@ static bool pb_field_set_to_default(pb_field_iter_t *field)
     }
     else if (PB_ATYPE(type) == PB_ATYPE_POINTER)
     {
+#ifdef PB_ENABLE_MALLOC
         /* Initialize the pointer to NULL. */
         *(void**)field->pField = NULL;
 
@@ -927,6 +935,7 @@ static bool pb_field_set_to_default(pb_field_iter_t *field)
         {
             *(pb_size_t*)field->pSize = 0;
         }
+#endif
     }
     else if (PB_ATYPE(type) == PB_ATYPE_CALLBACK)
     {
@@ -1494,7 +1503,10 @@ static bool checkreturn pb_dec_bytes(pb_istream_t *stream, const pb_field_iter_t
     if (PB_ATYPE(field->type) == PB_ATYPE_POINTER)
     {
 #ifndef PB_ENABLE_MALLOC
-        PB_RETURN_ERROR(stream, "no malloc support");
+        pb_size_t bytes_size = *(pb_size_t*)field->pField;
+        if (alloc_size > bytes_size)
+            PB_RETURN_ERROR(stream, "bytes overflow");
+        dest = (pb_bytes_array_t*)field->pData;
 #else
         if (stream->bytes_left < size)
             PB_RETURN_ERROR(stream, "end-of-stream");
```

Does this kind of feature fit the vision of nanopb? Is this something that would be reasonable to implement?
Maybe the preference would be to create a new type instead of `FT_POINTER`? I'd be curious to hear your thoughts.

PetteriAimonen, 2022-05-24:
I feel that this is a bit too specific to include in the main codebase. The use case makes sense, but e.g. strings do not have a size field that could be directly used in this way.

However, I think the name-bound callback support could make this just as easy without modifications to nanopb code.
Something like this (untested code):

```
message CommonMessage {
  uint64 time        = 1;
  uint32 message_id  = 2;
  bytes message_data = 3 [(nanopb).callback_datatype = "pb_bytes_array_t*",
                          (nanopb).callback_function = "callbackfunc_bytes_ptr"];
}
```

```
bool callbackfunc_bytes_ptr(pb_istream_t *istream, pb_ostream_t *ostream, const pb_field_iter_t *field)
{
    pb_bytes_array_t *data = *(pb_bytes_array_t**)field->pData
    if (istream)
    {
        /* Decode into preallocated bytes array */
        pb_istream_t substream;
        if (!pb_make_string_substream(istream, &substream)) return false;
        if (substream.bytes_left > data->size) PB_RETURN_ERROR(istream, "bytes exceed pre-allocated pointer field");
        return pb_read(&substream, data->bytes, substream.bytes_left)
            && pb_close_string_substream(istream, &substream);
    }
    else
    {
        /* Encode from preallocated array */
        return pb_encode_tag_for_field(ostream, field)
            && pb_encode_string(ostream, data->bytes, data->size);
    }
}
```

Using name-bound callback for this avoids the need to specify callback function pointers at invocation, and also allows customizing the field type in the struct as `pb_bytes_array_t*`.

Jonathan-Povirk-MSASafety, 2022-05-24:
Thank you for your input and the practical example, I'll investigate the callback route more thoroughly. I'll close as won't fix.

Jonathan-Povirk-MSASafety, 2022-05-24:
This particular feature does not fit with the vision of nanopb at this time. Equivalent functionality should be possible through the callback interface.