nanopb/nanopb #783: Could not find enum type when using mangle_names:M_STRIP_PACKAGE

PetteriAimonen, 2022-06-06:
From https://stackoverflow.com/questions/72520392/nanopb-strip-package-option-raises-error-at-import

file_A.proto:

```
syntax = "proto2";
package folder_A;
enum my_enum {
    ENUM_0 = 0;
    ENUM_1 = 1;
    ENUM_2 = 2;
}
```

file_A.options:

`* mangle_names:M_STRIP_PACKAGE`

In folder_B, I have file_B.proto :
```

syntax = "proto2";
package folder_B;
import "folder_A/file_A.proto";
message dummy  {
    required folder_A.my_enum value  = 1;
}

```
I try to generate proto file with the following command:

`nanopb_generator.py -D . -I . -I .\folder_B\ .\folder_A\file_A.proto .\folder_B\file_B.proto`

The script fails with error Exception: Could not find enum type folder_A_my_enum while generating default values for folder_B_dummy.

PetteriAimonen, 2022-06-06:
Affects also other uses of `M_STRIP_PACKAGE` when there are imports from another package.

Seems somewhat complex to handle correctly at this time, because the name mangling used in parsing the other file is not known when resolving the references.

RichardLangFromNZ, 2022-09-08:
There's a similar(/related) problem when trying to import a namespace-enclosed message definition from one proto file to use in a sub-message declaration in another file when a `mangle_names` option is provided.

Problem encountered with nanopb-0.4.5

bar.proto

```
syntax = "proto3";
import "nanopb.proto";

package bar_namespace;

option (nanopb_fileopt).mangle_names = M_STRIP_PACKAGE;

message BarMessage
{
}
```

foo.proto

```
syntax = "proto3";
import "bar.proto";
import "nanopb.proto";

package foo_namespace;

option (nanopb_fileopt).mangle_names = M_STRIP_PACKAGE;

message FooMessage
{
    bar_namespace.BarMessage bar_message = 1;
}
```

In this case the `protoc` compiler runs without error, however in the generated header files

bar.pb.h

```
...
/* Struct definitions */
typedef struct _BarMessage { 
    char dummy_field;
} BarMessage;
...
```

foo.pb.h

```
...
/* Struct definitions */
typedef struct _FooMessage { 
    bool has_bar_message;
    bar_namespace_BarMessage bar_message; 
} FooMessage;
...
```

the nanopb `mangle_names` option has been honored in the `BarMessage` type declaration in "bar.pb.h", but not in it's usage in "foo.pb.h".

closed 2022-12-11 by PetteriAimonen