nanopb/nanopb #785: Catastrophic failure on executing NANOPB_GENERATE_CPP() CMake function on ubuntu host machine with system install of nanopb

RichardLangFromNZ, 2022-06-15:
So, I incorporated nanopb into a CMake project as an in-source tree submodule, and used NANOPB_GENERATE_CPP() to compile my .proto files.  All worked as expected.

Then a colleague cloned my project and attempted to build it on an ubuntu host with a system package manager install of nanopb.   

The

```
find_path(NANOPB_GENERATOR_SOURCE_DIR
    NAMES nanopb_generator.py  
    DOC "nanopb generator source"  
    PATHS  
    ${NANOPB_SRC_ROOT_FOLDER}/generator  
    NO_CMAKE_FIND_ROOT_PATH  
)
```
search picks up "nanopb_generator.py" from "/usr/bin", then the 
```
  add_custom_command(
      OUTPUT ${NANOPB_GENERATOR_EXECUTABLE} ${GENERATOR_CORE_SRC}
      COMMAND ${CMAKE_COMMAND} -E copy_directory
      ARGS ${NANOPB_GENERATOR_SOURCE_DIR} ${GENERATOR_PATH}
      VERBATIM)
```
statement proceeds to copy the entire contents of "/usr/bin" into the build directory.

I think the find_path() call needs a NO_DEFAULT_PATH modifier.

RichardLangFromNZ, 2022-06-15:
Related to more general problems described in [Issue 481](https://github.com/nanopb/nanopb/issues/481)

RichardLangFromNZ, 2022-06-16:
As a short term stand-alone fix, changing all the find_xxx() function calls in "FindNanopb.cmake" to specify paths with HINTS directives (rather than PATHS) ensures that the suggested paths are searched first, such that the local versions of the desired items are found, even if there is a system install present

PetteriAimonen, 2022-06-16:
The copying of the files to build directory is annoying and mostly unnecessary anyway.
It got added to the CMake rules because someone got annoyed about the autogenerated nanopb_pb2.py in source directory.

In the ideal case the `nanopb_pb2.py` wouldn't need to be saved to filesystem at all.
There was work started on this in #601, and the [dev_dynamic_nanopb_pb2](https://github.com/nanopb/nanopb/compare/dev_dynamic_nanopb_pb2) branch seems to work quite ok, it would just need rebasing & more testing on different platforms.

PetteriAimonen, 2022-12-11:
Version 0.4.7 adds environment variable `NANOPB_PB2_TEMP_DIR`. Setting it to `1` will store `nanopb_pb2.py` to a system-wide temp folder, or it can be set to a path.

Using this should avoid need of copying the generator folder to build folder, so that `FindNanopb.cmake` could be simplified.

PetteriAimonen, 2023-11-11:
I'm not sure if this is still an issue in 0.4.8, but it seems the interoperability of FindNanopb.cmake and the CMakeLists.txt installation has not been designed or tested at all. There is a separate issue #911 about that.

PetteriAimonen, 2024-09-19:
I'm closing this off as I don't have a way to reproduce the problem and I think it may have been fixed by the numerous CMake build rule changes.

closed 2024-09-19 by PetteriAimonen