nanopb/nanopb #786: Issue encoding uint32 field

jacobw56, 2022-06-22:
I'm worried that this is some sort of oversight on my part, so I apologize in advance if this is me being silly.  I have a pretty simple proto for some sensor data:

    syntax = "proto3";

    message SensorData {
        double x_axis = 1;
        double y_axis = 2;
        double z_axis = 3;
        uint32 sensor_type = 4;
        uint32 seq = 5;
        uint32 stamp = 6;
        uint32 device = 7;
        uint32 batch = 8;
    }

Whenever I encode data with nanopb, I get something that looks like this:

    09 54 1a 31 b3 cf a3 11 40 11 8d 7d c9 c6 83 2d cc 3f 19 ed d4 5c 6e 30 cc 20 c0 28 d6 0a 30 f2 68 38 b4 a4 b4 de 0a 40 a7 e9 b6 c5 0b 00 00 00 dd fe 02 00 89 2a 03 00 bc

If you take a look at this using either [`protobufjs`](https://github.com/protobufjs/protobuf.js/) or [`protobuf-decoder`](https://github.com/pawitp/protobuf-decoder), which can be used [here](https://protobuf-decoder.netlify.app/), I can never recover field 4.  Attempting to decode this using [`protoc`](https://github.com/protocolbuffers/protobuf) simply fails with `Failed to parse input.`:

    $ protoc --decode_raw < proto_data_1.raw
    Failed to parse input.
    
    $ protoc --decode SensorData proto_def.proto < proto_data_1.raw
    Failed to parse input.

I tried changing the uint32 fields to fixed32 thinking that maybe it was something in packing variable integer data, but it only gets worse.  If I encode the same data with, for instance, `protobufjs`, I get this:

    09 54 1a 31 b3 cf a3 11 40 11 8d 7d c9 c6 83 2d cc 3f 19 ed d4 5c 6e 30 cc 20 c0 20 00 28 d6 0a 30 f2 68 38 b4 a4 b4 de 0a 40 a7 e9 b6 c5 0b

which I can now decode successfully with `protobufjs` and with `protoc`:

    $ protoc --decode_raw < proto_data_2.raw
    1: 0x4011a3cfb3311a54
    2: 0x3fcc2d83c6c97d8d
    3: 0xc020cc306e5cd4ed
    4: 0
    5: 1366
    6: 13426
    7: 2882343476
    8: 3098391719
    
    $ protoc --decode SensorData proto_def.proto < proto_data_2.raw
    x_axis: 4.409972
    y_axis: 0.220139
    z_axis: -8.398807
    seq: 1366
    stamp: 13426
    device: 2882343476
    batch: 3098391719

The reason I think this might be a me-problem is that you can see that even `protoc` doesn't seem to be in love with field 4, leading me to think that there is something inherently wrong with my proto to begin with, although I cannot for the life of me see it.

jacobw56, 2022-06-22:
In typical fashion I spoke too soon. Clearly I used `SensorData_size` when I sent the data instead of `pb_stream.bytes_written` and was seeing extra characters.  The `sensor_type` field is a red herring - it just isn't showing up because it looks like it has length zero (since it only contains 0x00).  Move along, nothing to see here.

PetteriAimonen, 2022-06-22:
Nice to hear you got it solved :)

closed 2022-06-22 by PetteriAimonen