nanopb/nanopb #787: Build issue with protobuf v4.21.0

pdgendt, 2022-06-27:
Updating the python package `protobuf` to v4.21.0 results in a build error for the simple example:

```
/home/pdgendt/dev/nanopb/examples/simple$ make
../../generator/protoc  --nanopb_out=. simple.proto
Traceback (most recent call last):
  File "/home/pdgendt/dev/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pdgendt/dev/nanopb/examples/simple/../../generator/protoc", line 9, in <module>
    from nanopb_generator import invoke_protoc
  File "/home/pdgendt/dev/nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/pdgendt/dev/nanopb/generator/proto/nanopb_pb2.py", line 34, in <module>
    _descriptor.EnumValueDescriptor(
  File "/home/pdgendt/.pyenv/versions/nanopbenv/lib/python3.9/site-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
make: *** [Makefile:21: simple.pb.c] Error 1
```

Both workarounds seem to help, but a proper fix would be preferred.

PetteriAimonen, 2022-06-27:
I think your `protoc` version maybe older than your `python-protobuf` version.

Try `pip3 install --upgrade grpcio_tools

pdgendt, 2022-06-28:
Yes, I used a distro `protoc`, version 3.6.1 (Ubuntu 20.04). However installing `grpcio_tools` also downgrades the `python-protobuf` to version 3.x.

shreyasbharath, 2022-08-13:
I am running in to a similar error, any suggestions? I tried `pip3 install --upgrade grpcio_tools` but that didn't work.

```
Traceback (most recent call last):
  File "/Users/shreyas/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package
```

pdgendt, 2022-08-16:
> I am running in to a similar error, any suggestions? I tried `pip3 install --upgrade grpcio_tools` but that didn't work.

Not sure if this helps, but it could be that a newer version of `protobuf` is already installed, you could try (use with caution/virtual environment):
```shell
pip3 install --upgrade --force-reinstall grpcio_tools
```

PetteriAimonen, 2022-08-16:
It might even make sense for nanopb generator to use `PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python` environment variable by default if it helps. After all speed is not that critical in the generator.

brentru, 2022-08-18:
Experiencing the same issue as @pdgendt. We patched it by setting our Action CI's env variable to `PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python` ([here](https://github.com/adafruit/Wippersnapper_Protobuf/commit/a34d4866ef42b8217e1ab5fbbbd8e898ca6e9194)).

@PetteriAimonen +1 for using the env. variable by default within the generator file.

pdgendt, 2022-10-06:
@PetteriAimonen this seems to be broken again with the latest version of `grpcio_tools` (v1.491).

pdgendt, 2022-10-06:
> If `PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python` works it could be set by default.

Yes, it does. Can this be the default for all build systems? Like `cmake` for example.

PetteriAimonen, 2022-10-06:
@pdgendt It is now default. Also available in pip/PyPi as prerelease 0.4.7.dev1355.

pdgendt, 2022-10-10:
@PetteriAimonen When is 0.4.7 expected to be released?

PetteriAimonen, 2022-10-11:
Your protoc version is probably older than your python-protobuf.
Try pip3 install --upgrade grpcio_tools

pdgendt, 2022-10-11:
> Your protoc version is probably older than your python-protobuf. Try pip3 install --upgrade grpcio_tools

```
$ pip3 install --upgrade grpcio_tools
Requirement already satisfied: grpcio_tools in /home/pdgendt/.pyenv/versions/3.10.5/envs/zephyr-test/lib/python3.10/site-packages (1.49.1)
Requirement already satisfied: setuptools in /home/pdgendt/.pyenv/versions/3.10.5/envs/zephyr-test/lib/python3.10/site-packages (from grpcio_tools) (65.4.1)
Requirement already satisfied: protobuf<5.0dev,>=4.21.3 in /home/pdgendt/.pyenv/versions/3.10.5/envs/zephyr-test/lib/python3.10/site-packages (from grpcio_tools) (4.21.7)
Requirement already satisfied: grpcio>=1.49.1 in /home/pdgendt/.pyenv/versions/3.10.5/envs/zephyr-test/lib/python3.10/site-packages (from grpcio_tools) (1.49.1)
Requirement already satisfied: six>=1.5.2 in /home/pdgendt/.pyenv/versions/3.10.5/envs/zephyr-test/lib/python3.10/site-packages (from grpcio>=1.49.1->grpcio_tools) (1.16.0)
```

closed 2022-12-11 by PetteriAimonen