nanopb/nanopb #793: criterion fails to link with libprotobuf-nanopb.so: ld: error: undefined symbol: pb_release

yurivict, 2022-07-03:
https://github.com/Snaipe/Criterion

```
cc  -o src/libcriterion.so.3.2.0 src/libcriterion.so.3.2.0.p/meson-generated_.._criterion.pb.c.o src/libcriterion.so.3.2.0.p/err.c.o src/libcriterion.so.3.2.0.p/mutex.c.o src/libcriterion.so.3.2.0.p/capi_specifiers.c.o src/libcriterion.so.3.2.0.p/capi_stream.c.o src/libcriterion.so.3.2.0.p/compat_alloc.c.o src/libcriterion.so.3.2.0.p/compat_kill.c.o src/libcriterion.so.3.2.0.p/compat_mockfile.c.o src/libcriterion.so.3.2.0.p/compat_path.c.o src/libcriterion.so.3.2.0.p/compat_pipe.c.o src/libcriterion.so.3.2.0.p/compat_process.c.o src/libcriterion.so.3.2.0.p/compat_processor.c.o src/libcriterion.so.3.2.0.p/compat_section.c.o src/libcriterion.so.3.2.0.p/compat_strtok.c.o src/libcriterion.so.3.2.0.p/compat_time.c.o src/libcriterion.so.3.2.0.p/core_abort.c.o src/libcriterion.so.3.2.0.p/core_assert.c.o src/libcriterion.so.3.2.0.p/core_client.c.o src/libcriterion.so.3.2.0.p/core_ordered-set.c.o src/libcriterion.so.3.2.0.p/core_report.c.o src/libcriterion.so.3.2.0.p/core_runner.c.o src/libcriterion.so.3.2.0.p/core_runner_coroutine.c.o src/libcriterion.so.3.2.0.p/core_stats.c.o src/libcriterion.so.3.2.0.p/core_test.c.o src/libcriterion.so.3.2.0.p/csptr_array.c.o src/libcriterion.so.3.2.0.p/csptr_mman.c.o src/libcriterion.so.3.2.0.p/entry_entry.c.o src/libcriterion.so.3.2.0.p/entry_options.c.o src/libcriterion.so.3.2.0.p/entry_params.c.o src/libcriterion.so.3.2.0.p/io_event.c.o src/libcriterion.so.3.2.0.p/io_file.c.o src/libcriterion.so.3.2.0.p/io_json.c.o src/libcriterion.so.3.2.0.p/io_output.c.o src/libcriterion.so.3.2.0.p/io_redirect.c.o src/libcriterion.so.3.2.0.p/io_tap.c.o src/libcriterion.so.3.2.0.p/io_xml.c.o src/libcriterion.so.3.2.0.p/log_logging.c.o src/libcriterion.so.3.2.0.p/log_normal.c.o src/libcriterion.so.3.2.0.p/protocol_connect.c.o src/libcriterion.so.3.2.0.p/protocol_messages.c.o src/libcriterion.so.3.2.0.p/protocol_protocol.c.o src/libcriterion.so.3.2.0.p/string_brz.c.o src/libcriterion.so.3.2.0.p/string_extglobmatch.c.o src/libcriterion.so.3.2.0.p/string_fmt.c.o src/libcriterion.so.3.2.0.p/string_i18n.c.o src/libcriterion.so.3.2.0.p/string_xxd.c.o src/libcriterion.so.3.2.0.p/string_diff.c.o src/libcriterion.so.3.2.0.p/core_theories.c.o -Wl,--as-needed -Wl,--no-undefined -Wl,-O1 -shared -fPIC -Wl,--start-group -Wl,-soname,libcriterion.so.3 -fstack-protector-strong -O2 -pipe -fno-omit-frame-pointer -fPIC -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -Wl,-rpath,/usr/local/lib -Wl,-rpath-link,/usr/local/lib -Wl,--exclude-libs,ALL -pthread /usr/local/lib/libboxfort.so /usr/local/lib/libffi.so /usr/local/lib/libgit2.so /usr/local/lib/libnanomsg.so /usr/local/lib/libprotobuf-nanopb.so.0 -lrt -lm -Wl,--end-group
ld: error: undefined symbol: pb_release
>>> referenced by messages.c
>>>               src/libcriterion.so.3.2.0.p/protocol_messages.c.o:(cr_send_to_runner)
>>> referenced by messages.c
>>>               src/libcriterion.so.3.2.0.p/protocol_messages.c.o:(free_message)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
```

Is this symbol missing in ```libprotobuf-nanopb.so``` by any chance?

PetteriAimonen, 2022-07-03:
That sounds like it expects a library built with `PB_ENABLE_MALLOC` defined.

yurivict, 2022-07-03:
> That sounds like it expects a library built with PB_ENABLE_MALLOC defined.

How to achieve this through cmake build?

yurivict, 2022-07-03:
With PB_ENABLE_MALLOC defined Crterion indeed succeeds.
You probably want to enable it by default?

PetteriAimonen, 2022-07-03:
No, I don't. The primary target platform of nanopb is embedded systems, where malloc is generally avoided.

This is probably something that should be resolved in the build scripts / instructions of Crterion.

yurivict, 2022-07-03:
Perhaps a cmake option defaulting to OFF with explanation is the right place?

Et7f3, 2022-08-12:
I don't know why you are getting issue ? https://github.com/Snaipe/Criterion/blob/f6097bcf699e436e4033e10e2a5f0e13d1b46e92/meson.build#L181

PetteriAimonen, 2022-11-25:
I'm closing this as fixed, please reopen if there is still some issue that I can do something about.

closed 2022-11-25 by PetteriAimonen