nanopb/nanopb #795: "mangle_names" option not parsed properly by CMake nanopb_generate_cpp() script

RichardLangFromNZ, 2022-07-14:
Using NanoPb as in-source submodule from commit hash 529d26b (01-May-2022)

CMakeLists.txt script
```
...
find_package(Nanopb REQUIRED)
...
set(NANOPB_OPTIONS "-s mangle_names:M_STRIP_PACKAGE")

nanopb_generate_cpp(LIB_NANOPB_SRC_FILES LIB_NANOPB_HDR_FILES ${PROJECT_PROTO_SRC_FILES})
...
```
results in the following exception stack from "nanopb_generator.py" during the build phase

```
...
[build] Traceback (most recent call last):
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2338, in <module>
[build]     main_plugin()
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2315, in main_plugin
[build]     other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2113, in parse_file
[build]     f = ProtoFile(fdesc, file_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1563, in __init__
[build]     self.parse()
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1648, in parse
[build]     self.messages.append(Message(name, message, message_options, index, self.comment_locations))
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1142, in __init__
[build]     self.load_fields(desc, message_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1179, in load_fields
[build]     field = Field(self.name, f, field_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 583, in __init__
[build]     self.ctype = names_from_type_name(desc.type_name)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 195, in names_from_type_name
[build]     raise NotImplementedError("Lookup of non-absolute type names is not supported")
[build] NotImplementedError: Lookup of non-absolute type names is not supported
[build] Traceback (most recent call last):
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2338, in <module>
[build]     main_plugin()
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2315, in main_plugin
[build]     other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 2113, in parse_file
[build]     f = ProtoFile(fdesc, file_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1563, in __init__
[build]     self.parse()
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1648, in parse
[build]     self.messages.append(Message(name, message, message_options, index, self.comment_locations))
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1142, in __init__
[build]     self.load_fields(desc, message_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 1179, in load_fields
[build]     field = Field(self.name, f, field_options)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 583, in __init__
[build]     self.ctype = names_from_type_name(desc.type_name)
[build]   File "/home/richard/Professional/RPL/develop/firmware/ugv/PMB/UGV-power-management-board/build/proto-ipc/nanopb/generator/nanopb_generator.py", line 195, in names_from_type_name
[build]     raise NotImplementedError("Lookup of non-absolute type names is not supported")
[build] NotImplementedError: Lookup of non-absolute type names is not supported
[build] --nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
...
```

for comparison, a `set(NANOPB_OPTIONS "-s long_names:false")` option is applied as expected

PetteriAimonen, 2022-07-14:
Can you include the `.proto` file that reproduces the problem?

Does it only occur through CMake, or also when calling `nanopb_generator.py -s mangle_names:M_STRIP_PACKAGE myproto.proto` directly?

RichardLangFromNZ, 2022-07-17:
[pmb_topics.proto.zip](https://github.com/nanopb/nanopb/files/9128529/pmb_topics.proto.zip)

Yep, same problem calling `nanopb_generator.py `directly.  I can specify a `-s long_names:false` option and it works as expected, but `-s mangle_names:M_STRIP_PACKAGE` falls over

PetteriAimonen, 2022-11-25:
Looks like specifying the same option inside the file works for some reason:

    option (nanopb_fileopt).mangle_names = M_STRIP_PACKAGE;

PetteriAimonen, 2022-11-25:
It was caused by the `mangle_names` option getting applied to the imported `nanopb.proto` also, and generator not handling it because there was no package name.

Should be fixed now.

closed 2022-12-11 by PetteriAimonen