nanopb/nanopb #796: typedef support for nanopb

Siddhant085, 2022-07-21:
I am experimenting with using nanopb to provide a backward and forward compatible linux kernel interface. I will be porting my data structures from C to proto. C supports typedef that enables us to define custom types which can be used in other data structures. This improves the readability and simplifies the structure of the code.

Most protocol buffer implementations don't support typedef with the argument that protocol buffers are designed to be language independent and thus don't want to introduce language specific features. However this is not the case for nanopb, which has been designed specially for C and has added support for language specific features using the nanopb.options file.

Example use case:

In C we often typedef uint64 as size_t and use it buffers like so:
```
typedef uint64 size_t

struct buffer {
    size_t size;
    char *data;
}
```

This kind be enabled in nanopb using nanopb.options like so

```
import "nanopb.proto"
message size_t {
    required uint64 value = 1;
    (nanopb).isTypedef = true;
}
```

We need to change the code generation fro message to check if the flag has been enabled and generate the typedef instead of defining the structure.

PetteriAimonen, 2022-07-21:
Adding submessages unnecessarily changes the encoding of the data. So it might be better to have something like:

```
option (nanopb_fileopt).include = "my_typedefs.h";

message MyMessage {
       uint64 offset = 1 [(nanopb).type_name = "size_t"];
}
```

which would then generate:

```
#include "my_typedefs.h"

typedef struct _MyMessage {
    size_t offset;
} MyMessage;

#define MyMessage_FIELDLIST(X, a) \
X(a, STATIC,   REQUIRED, UINT64,    offset,         1)

```

That would allow retaining message type compatibility with other libraries without introducing unnecessary submessages.
It also allows compile-time variation in `my_typedefs.h` to account e.g. 32 vs. 64 bit platform.

Siddhant085, 2022-07-21:
This is interesting. Using this way we can also support predefined types which might be included from some C libraries.
However won't this place the responsibility of finding the right data type (uint64 in the example you showed) on the user. Also since the user will be doing it manually it won't automatically pick that up from my_typedefs.h

I am trying to understand the encoding process from the code and the message structure will be generated based on the uint64 value which should match the exact datatype in my_typedefs.h. Please correct me if I am misinterpreting something.

djacobowitz-cmt, 2022-07-22:
Is the goal just to provide aliases for base types that nanopb already understands? If so, maybe the most elegant solution is to have some directive that tells nanopb to use treat X in a proto file as a Y, and to use X in the generated headers.

PetteriAimonen, 2022-07-22:
The type in `FIELDLIST` define will determine which encoder/decoder function gets invoked, through this mapping:
https://github.com/nanopb/nanopb/blob/master/pb.h#L843-L864

As long as the substituted type maps to the same LTYPE, it can be easily replaced in the struct. The size of varint integers can be different, it is determined at compile time using `sizeof()`.

Siddhant085, 2022-07-25:
So in case where the nanopb.typedef option is specified we change the output to produce a field with type mentioned in this option, while we set the LTYPE to the data type we want to use for encoding. Finding the correct datatype can be difficult as there can be nested typedefs

```
#define uint64 size_t
#define size_t buffer_size_t
#define buffer_size_t string_size_t
```

For the above example we will need to keep track of the dependency structure of the typedef and replace the LTYPE with the base type that is uint64. Typedef can be used for struct also so the LTYPE can be PB_LTYPE_MAP_MESSAGE as well, though I don't think that should be a problem. Moreover with this approach we lose the capability of using types defined in some other header like "my_typedefs.h". So we can't automatically accommodate for compile time variations to account for int32 and int64

PetteriAimonen, 2022-07-26:
int32 and int64 are the same LTYPE, so there is no need to FIELDLIST for that specific pair.

Changing protobuf encoding types based on some C typedefs does not make sense to me.
Protobuf encoding stays compatible exactly when the LTYPE remains the same.
The main point of protobuf is to have the portable `.proto` file as interface specification.
If you ditch that, you might as well just transfer raw structs.

Siddhant085, 2022-07-26:
Ok, that makes sense. So the final idea is to change the LTYPE of the field based on the typedef and for this we need to maintain a tree of dependencies which will tell us what the LTYPE for any typedef should be.

PetteriAimonen, 2022-07-26:
No, I do not think changing LTYPE based on typedef makes any sense.

LTYPE must always match the type in .proto, otherwise the encoding doesn't have any chance of interoperability.

Siddhant085, 2022-07-26:
Ok, I am getting confused between the original and this new idea.

Original idea was that the typedef will be defined as a message, which will be written out as typedef declaration and then the other messages can use it as a type. Example:
```
import "nanopb.proto"
message size_t {
    required uint64 value = 1;
    (nanopb).isTypedef = true;
}
message buffer {
    required size_t size = 1 [(nanopb).type_name = "size_t"];;
    required string data = 2;
}
```
The above proto generating:
```
typedef uint64 size_t

struct buffer {
    size_t size;
    char *data;
}
```
The new idea is instead of using submessages we want the fields to be defined as typedef and change the LTYPE based on the type of the typedef. **The important point being that the typedef is defined in the proto file itself so that it stays interoperable**.

```
message size_t {
    required uint64 value = 1;
    (nanopb).isTypedef = true;
}
message buffer {
    required size_t size = 1;
    required string data = 2;
}
```
But instead of replacing size in buffer with a submessage we change the LTYPE of size based on the LTYPE of value in size_t, while writing out size_t as the type in the pb.h file.

PetteriAimonen, 2022-07-26:
Ok, I obviously do not understand your idea.

The example usecase seems very weird. Why `size_t` as a submessage? Why data as `string` instead of `bytes`? Why size separately when it is already part of bytes?

Siddhant085, 2022-07-26:
My idea is to provide the feature of defining custom types using typedef for clarity and simplicity in the C code being generated. All the arguments supporting the existence of typedef in C can be applied here.

My example might have been a little confusing. Yes it makes more sense to use bytes in the example I described but that is only for this particular example. Let me give another example.

Assume we have a queue which has a queue ID. This ID field is of type `uint32` and the ID is being used in multiple structures. Now if we define the type as `uint32` we will define the ID as `uint32 queue_id` everywhere we use the ID. Lets say in the future we want to change the size of `queue_id` to `uint64`, now we will have to look through all the uses of this `queue_id` and replace each one of them with `uint64`. Note that the variable name might be different making it a lot harder. Now if we had typedef we can do `typedef uint32 queue_id_t` and then use `queue_it_t queue_id` wherever we want a variable to store the queue id.  Now to change from uint32 to unit64 we simply change the typedef to `typedef uint64 queue_it_t` and we are done.

PetteriAimonen, 2022-07-26:
For such an use, I would just typedef it in C, and if needed, make a static assert that the size generated from `.proto` is equivalent.

Siddhant085, 2022-07-26:
That will work but that will be a workaround and not a very clean solution.
Can you help me understand, the problem with supporting it in nanopb? You mentioned that changing the LTYPE can mess with encoding/decoding process. Can you expand more on that and mention the concerns you have with supporting typedef in C?

PetteriAimonen, 2022-07-27:
The numeric LTYPE value (which it gets through the `#define` mappings) directly defines the format the field gets in encoded message. Changing this to anything other than the value defined in `.proto` file will break compatibility with other protobuf libraries, which I do not want.

In any case, I have decided that the presented use cases are not important enough to warrant the inclusion of this feature in main nanopb codebase.

closed 2022-07-27 by PetteriAimonen