nanopb/nanopb #800: what difference does nanopb make?

pippy360, 2022-08-01:
Hi,

The homepage/readmes don't seem to make it clear what difference nanopb makes?

From what I can tell it gives us a full protobuf library but it's considerably smaller than the normal protobuf library so we take up  less ram? but how much smaller is it? and are there any trade offs for this size reduction? I would love if the documentation gave some indication what difference using nanopb will make.

I ask because I've been asked to support a library which is using nanopb and I can't figure out why we decided to use it and if it's still useful for us.

Thanks for all the work,
Tom

PetteriAimonen, 2022-08-01:
The main differences are listed here: https://jpa.kapsi.fi/nanopb/docs/index.html#features-and-limitations

I'm not sure what is the "normal protobuf library" you refer to. I assume Google's protobuf C++ library?
They are in quite different size category.
I haven't used Google's C++ library on embedded systems, but on my x86_64 system `libprotobuf` is 3 megabytes of code, `libprotobuf-lite` is 461 kB and nanopb is 21 kB with malloc support and 18 kB without.

Overall the nanopb API is quite different from Google's C++ library, because the latter very heavily depends on dynamic allocation and C++ features. Might be better for some purposes and worse for others.

I know that some people prefer to use nanopb on PC also for various reasons, but the primary target platform has always been microcontrollers, which have no hope of fitting the protobuf C++ library.

pippy360, 2022-08-01:
That's a really helpful reply. Thank you

> I assume Google's protobuf C++ library?
yes

closed 2022-11-24 by PetteriAimonen