nanopb/nanopb #801: Cannot install nanopb 0.4.6.4 on PlatformIO

NorbertSandor, 2022-08-01:
The latest VSCode+PIO environment does not seem to like the version number of the latest versions: `nanopb/Nanopb@^0.4.6.4`

Error message: `ValueError: Invalid simple block '^0.4.6.4'`

Stacktrace:
```
Library Manager: Removing Nanopb @ 0.4.6+4
Library Manager: Nanopb@0.4.6+4 has been removed!
Error: Traceback (most recent call last):
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 144, in cli
    process_env(
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 201, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\run\processor.py", line 83, in process
    install_project_env_dependencies(
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 241, in _install_project_env_libraries
    spec = PackageSpec(library)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 184, in __init__
    self._parse(self.raw)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 291, in _parse
    raw = parser(raw)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 316, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 231, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\Norbi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '^0.4.6.4'
```

PetteriAimonen, 2022-08-01:
From the messages, it seems that platformio wants to call it `0.4.6+4`.

PlatformIO is strongly assuming semantic versioning, but nanopb version numbering is not currently that. I'll probably change it some time.

NorbertSandor, 2022-08-03:
Maybe it would be an easy fix if the next release would use x.y.z version number format.

PetteriAimonen, 2022-08-04:
Yes, that is what I usually do. The reason why for the `0.4.6.4` was a mistake, `0.4.6` got pulled to platform.io registry too early because a pull request bumped the version number and I did not notice. And there is apparently no way to go back on versions.

maxgerhardt, 2022-08-10:
Tracked in https://github.com/platformio/platformio-core/issues/4389.

NorbertSandor, 2022-09-09:
> selecting version 0.4.6 instead of 0.4.6.4 downloads it correctly

Yes, but it is not the latest version.

PetteriAimonen, 2022-09-09:
@NorbertSandor Hmm, can you provide more details on that, log and platformio.ini?

For me all of these install the latest version, 0.4.6.4 which platformio just calls 0.4.6+4:

    lib_deps =
          nanopb/nanopb @ 0.4.6+4

    lib_deps =
          nanopb/nanopb @ 0.4.6

    lib_deps =
          nanopb/nanopb

Only specifying the version as `0.4.6.4` does not work and throws platformio error message.

You can confirm the installed version by checking `.pio/libdeps/template/Nanopb/library.json`

NorbertSandor, 2022-09-09:
You are right, the problem arise when you try to install using the platformio vscode UI:

[image]

[image]

PetteriAimonen, 2022-12-11:
This version number problem should now be history as 0.4.7 is published. I'll be more careful about numbering on platformio in the future.

closed 2022-12-11 by PetteriAimonen