nanopb/nanopb #802: pb_release definition if PB_ENABLE_MALLOC disables causes issues with cmock

eikehahn, 2022-08-02:
Hi,
we are using cmock as test framework in our project and since I updated nanopb from 0.4.1 to 0.4.6 we encountered an issue with `#define pb_release(fields, dest_struct) PB_UNUSED(fields); PB_UNUSED(dest_struct);` in pb_decode.h:118. A known cmock issue https://github.com/ThrowTheSwitch/CMock/issues/176 results in an uncompilable test framework.

The question is where is this pb_release defined for? I just removed it from our project, but since pb_release is also guarded in pb_decode.c, this line feels superfluous.

```
#ifdef PB_ENABLE_MALLOC
/* Release any allocated pointer fields. If you use dynamic allocation, you should
 * call this for any successfully decoded message when you are done with it. If
 * pb_decode() returns with an error, the message is already released.
 */
void pb_release(const pb_msgdesc_t *fields, void *dest_struct);
#else
/* Allocation is not supported, so release is no-op */
#define pb_release(fields, dest_struct) PB_UNUSED(fields); PB_UNUSED(dest_struct);
#endif
```

Best regards,
Eike

PetteriAimonen, 2022-08-02:
The purpose of the define is for code that calls `pb_release()` by convention every time a message is destroyed. For example C++ wrappers can do that. Having the function available always avoids needing to have guards everywhere.

The alternative definition could be moved to .c file instead, avoiding guards in header. Would that solve the problem for CMock?

eikehahn, 2022-08-02:
Just tested it and this would solve the problem for CMock, yes. Thank you for this fast reply :)

PetteriAimonen, 2022-08-02:
That change is simple enough.

However I now remembered there is a pull request #779 that would add a guard back:

```
#ifdef PB_ENABLE_MALLOC_CONTEXT
/* stream is optionally used for error messages and to set realloc function, if NULL then pb_free() is used */
void pb_release(pb_istream_t *stream, const pb_msgdesc_t *fields, void *dest_struct);
#else
void pb_release(const pb_msgdesc_t *fields, void *dest_struct);
#endif
```

It is a bit different, but I wonder if that will still result in uncompilable test set because it would result in a duplicate definition?
It is much harder to avoid this case.

eikehahn, 2022-08-02:
yes, I think this will make the test code uncompilable for the mentioned reason. Hopefully cmock will be fixed in the meanwhile.

PetteriAimonen, 2022-11-25:
Fixed the basic case now, will have to think about #779 more deeply once I get to it.

closed 2022-12-11 by PetteriAimonen